pyxem/pyxem #44: Docs warnings

thomasaarholt, 2017-07-12:
When building the docs, I get the following warnings:
Ignore the hyperspy ones.

```shell
C:\Users\thomasaar\Downloads\pycrystem-master\docs
(cryst) λ sphinx-build -b html source build
Running Sphinx v1.6.3
making output directory...
The fast cython based bcf unpacking library were found
WARNING: The ipywidgets GUI elements are not available, probably because the hyperspy_gui_ipywidgets package is not installed.
WARNING: The traitsui GUI elements are not available, probably because the hyperspy_gui_traitui package is not installed.

c:\programdata\miniconda3\envs\cryst\lib\site-packages\pymatgen\util\plotting_utils.py:3: UserWarning: pymatgen.util.plotting_utils has been renamed pymatgen.util.plotting. This stub will be removed in PMG 5.0.
  warnings.warn("pymatgen.util.plotting_utils has been renamed "
loading pickled environment... not yet created
loading intersphinx inventory from https://docs.python.org/objects.inv...
intersphinx inventory has moved: https://docs.python.org/objects.inv -> https://docs.python.org/2/objects.inv
loading intersphinx inventory from http://docs.scipy.org/doc/numpy/objects.inv...
intersphinx inventory has moved: http://docs.scipy.org/doc/numpy/objects.inv -> https://docs.scipy.org/doc/numpy/objects.inv
loading intersphinx inventory from http://hyperspy.org/hyperspy-doc/current/objects.inv...
loading intersphinx inventory from https://docs.scipy.org/doc/scipy/objects.inv...
WARNING: intersphinx inventory 'https://docs.scipy.org/doc/scipy/objects.inv' not fetchable due to <class 'requests.exceptions.HTTPError'>: 404 Client Error: Not Found for url: https://docs.scipy.org/doc/scipy/objects.inv
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 12 source files that are out of date
updating environment: 12 added, 0 changed, 0 removed
reading sources... [100%] strain_mapping
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\diffraction_generator.py:docstring of pycrystem.diffraction_generator.DiffractionSimulation.direct_beam_mask:1: WARNING: Inline interpreted text or phrase reference start-string without end-string.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\indexation_generator.py:docstring of pycrystem.indexation_generator.IndexationGenerator.correlate:6: WARNING: Inline strong start-string without end-string.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\indexation_generator.py:docstring of pycrystem.indexation_generator.IndexationGenerator.correlate:8: WARNING: Inline strong start-string without end-string.
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\indexation_generator.py:docstring of pycrystem.indexation_generator.MatchingResults.get_phase_results:4: (SEVERE/4) Unexpected section title.

Paramters
---------
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\docs\source\pycrystem.rst:65: (WARNING/2) autodoc: failed to import module 'pycrystem.orientation_map'; the following exception was raised:
Traceback (most recent call last):
  File "c:\programdata\miniconda3\envs\cryst\lib\site-packages\sphinx\ext\autodoc.py", line 657, in import_object
    __import__(self.modname)
ModuleNotFoundError: No module named 'pycrystem.orientation_map'
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\tensor_field.py:docstring of pycrystem.tensor_field.DisplacementGradientMap.polar_decomposition:3: WARNING: Unexpected indentation.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\expt_utils.py:docstring of pycrystem.utils.expt_utils.refine_beam_position:4: WARNING: Unexpected indentation.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\expt_utils.py:docstring of pycrystem.utils.expt_utils.refine_beam_position:5: WARNING: Block quote ends without a blank line; unexpected unindent.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\peakfinders2D.py:docstring of pycrystem.utils.peakfinders2D.find_peaks_stat:6: WARNING: Unexpected indentation.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\peakfinders2D.py:docstring of pycrystem.utils.peakfinders2D.find_peaks_stat:7: WARNING: Block quote ends without a blank line; unexpected unindent.
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\docs\source\pycrystem.utils.rst:50: (WARNING/2) autodoc: failed to import module 'pycrystem.utils.strain_utils'; the following exception was raised:
Traceback (most recent call last):
  File "c:\programdata\miniconda3\envs\cryst\lib\site-packages\sphinx\ext\autodoc.py", line 657, in import_object
    __import__(self.modname)
ModuleNotFoundError: No module named 'pycrystem.utils.strain_utils'
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:7: (SEVERE/4) Unexpected section title.

Parameters
----------
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:17: WARNING: Inline strong start-string without end-string.
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:19: WARNING: Definition list ends without a blank line; unexpected unindent.
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:20: (SEVERE/4) Unexpected section title.

Returns
-------
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:23: WARNING: Definition list ends without a blank line; unexpected unindent.
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:24: (SEVERE/4) Unexpected section title.

References
----------
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate:26: WARNING: Enumerated list ends without a blank line; unexpected unindent.
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate_component:9: (SEVERE/4) Unexpected section title.

Parameters
----------
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate_component:17: (SEVERE/4) Unexpected section title.

Returns
-------
WARNING: C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate_component:22: (SEVERE/4) Unexpected section title.

References
----------
C:\Users\thomasaar\Downloads\pycrystem-master\pycrystem\utils\__init__.py:docstring of pycrystem.utils.correlate_component:24: WARNING: Enumerated list ends without a blank line; unexpected unindent.
C:\Users\thomasaar\Downloads\pycrystem-master\docs\source\setup.rst:4: WARNING: autodoc: failed to import module 'setup'; the module executes module level statement and it might call sys.exit().
looking for now-outdated files... none found
pickling environment... done
checking consistency... C:\Users\thomasaar\Downloads\pycrystem-master\docs\source\roadmap.rst: WARNING: document isn't included in any toctree
done
preparing documents... done
writing output... [100%] strain_mapping
generating indices... genindex py-modindex
highlighting module code... [100%] pycrystem.utils
writing additional pages... search
copying static files... done
copying extra files... done
dumping search index in English (code: en) ... done
dumping object inventory... done
build succeeded, 27 warnings.
```

thomasaarholt, 2017-07-12:
That said, I see they are not particularly complete :)

dnjohnstone, 2017-08-09:
Docs are very much a work in progress - we'll get there soon hopefully! Maybe even next week.

closed 2018-01-08 by dnjohnstone