pyxem/pyxem #46: get_unique_vectors does not find all unique vectors AND distance_threshold solution proposed

tinabe, 2017-08-08:
_get_unique_vectors_ does not find all unique vectors. It considers each vector element x and y separately. If one of those is already in the gvlist, the considered vector will not be added, even if it is a unique vector. 
..................................................................

**Example of issue**

When testing the vector i[0][j] against the list of unique g-vectors, gvlist, the code will pass, meaning that this vector is not added to the list of unique vectors. The reason is that one of the values in i[0][j]; -0.11913, is in gvlist. However, this is not the same g-vector, and i[0][j] should have been added. 

i[0][j] = [ 0.26334 -0.11913] 
gvlist = [[ 0.35112 -0.11913]
..................................................................

**Solution**

A solution could be (using the same notation) to change the if-statement from: 

>` if np.asarray(i[0][j]) in np.asarray(gvlist): `

to 

`if sum(map(lambda x: np.array_equal(np.asarray(i[0][j]),x), np.asarray(gvlist))):`

This seems to works for me. Of course, since gvlist then becomes a lot larger, everything you do from there takes so much more time and space...

**Incorporating distance threshold**

Therefore, I tried to complete the to-do-task of adding the distance_threshold. I think this can be easily done using np.allclose. Then, instead change the above if-statement to: 

**`if sum(map(lambda x: np.allclose(np.asarray(i[0][j]), x, rtol=0, atol=distance_threshold, equal_nan=False), np.asarray(gvlist))):`** 

The complete function then becomes: 

```
    def get_unique_vectors(self,
                           distance_threshold=None):

        gvlist=[]

        for i in self._iterate_signal():

            for j in np.arange(len(i[0])):
                
                if distance_threshold is not None:
                    check = sum(map(lambda x: np.allclose(np.asarray(i[0][j]),
                                                          x, 
                                                          rtol=0, 
                                                          atol=distance_threshold, 
                                                          equal_nan=False), 
                                    np.asarray(gvlist)))
                        
                else: 
                     check = sum(map(lambda x: np.array_equal(np.asarray(i[0][j]),x), 
                                     np.asarray(gvlist)))
                                
                if check:
                    pass
                
                else:
                    gvlist.append(i[0][j])

        unique_vectors = np.asarray(gvlist)
        
        return unique_vectors
```

Hopefully, someone else has the time to test this as well.

dnjohnstone, 2017-08-09:
Hi @tinabe - thanks for this. Yes, that's a bug I'd been meaning to get to and just hadn't found time. Thanks!!

Please could you insert your bits of code into the source code (in a new branch on your machine) and make a pull request here? That might sound like a pain - but it will mean you get credit for your code and ultimately authorship in the future.

tinabe, 2017-08-10:
That's no problem, @dnjohnstone ! I tried to put it in a new branch and make a pull request as you suggested. I hope that is all right. :) I didn't add the suggestion above though, but a faster one using the map function.

dnjohnstone, 2017-08-17:
Resolved via #47

closed 2017-08-17 by dnjohnstone