pyxem/pyxem #57: Multislice forward model implementation

dnjohnstone, 2017-09-14:
We want to have some full dynamical diffraction simulations callable from PyCrystEM. Following e-mail correspondence the plan is to go for a multislice implementation and a Bloch wave implementation based on code from Christoph Koch's group. @woutervandenbroek and @jacobjma

The multislice code is here:
https://github.com/woutervandenbroek/FDES

A python wrapper for some of the desired functionality is here:
https://github.com/jacobjma/PyQSTEM/blob/master/examples/HRTEM_dft.ipynb

dnjohnstone, 2017-10-04:
@pc494 this is the multislice code that we'd like to wrap.

pc494, 2017-11-16:
This has been revised, it now looks likely we will be wrapping https://github.com/prism-em 

Currently waiting on a resolution to https://github.com/prism-em/prismatic/issues/6

closed 2019-04-27 by pc494