pyxem/pyxem #58: Bug in DiffractionSimulation.as_signal()

stefsmeets, 2017-09-28:
To reproduce:

    import pycrystem as pc
    ds = pc.diffraction_generator.DiffractionSimulation(coordinates = (np.random.random((10,2))))
    ds.as_signal(10, 3, 1.0)

Gives error:
    ImportError: can't import 'None' with 'import_module('numexpr')' command

Here is the full traceback:

    ImportError                               Traceback (most recent call last)
    <ipython-input-194-b24246395d7b> in <module>()
          1 ds = pc.diffraction_generator.DiffractionSimulation(coordinates = (np.random.random((10,2))))
    ----> 2 ds.as_signal(10, 3, 1.0)
    
    c:\users\stef\python\pycrystem\pycrystem\diffraction_generator.py in as_signal(self, size, sigma, max_r)
        276         x, y = np.meshgrid(l, l)
        277         coords = self.coordinates[:, :2]
    --> 278         g = Expression(_GAUSSIAN2D_EXPR, 'Gaussian2D', module='numexpr')
        279         for (cx, cy), intensity in zip(coords, self.intensities):
        280             g.intensity.value = intensity
    
    C:\Users\stef\AppData\Local\Continuum\Miniconda3\lib\site-packages\hyperspy\_components\expression.py in __init__(self,     expression, name, position, module, autodoc, add_rotation, rotation_center, **kwargs)
        119         self._str_expression = expression
        120         if rotation_center is None:
    --> 121             self.compile_function(module=module, position=position)
        122         else:
        123             self.compile_function(module=module, position=rotation_center)
    
    C:\Users\stef\AppData\Local\Continuum\Miniconda3\lib\site-packages\hyperspy\_components\expression.py in compile_function(    self, module, position)
        186         variables = [x, y] if self._is2D else [x]
        187         self._f = lambdify(variables + parameters, eval_expr,
    --> 188                            modules=module, dummify=False)
        189 
        190         if self._is2D:
    
    C:\Users\stef\AppData\Local\Continuum\Miniconda3\lib\site-packages\sympy\utilities\lambdify.py in lambdify(args, expr,     modules, printer, use_imps, dummify)
        377     namespace = {}
        378     for m in namespaces[::-1]:
    --> 379         buf = _get_namespace(m)
        380         namespace.update(buf)
        381 
    
    C:\Users\stef\AppData\Local\Continuum\Miniconda3\lib\site-packages\sympy\utilities\lambdify.py in _get_namespace(m)
        469     """
        470     if isinstance(m, str):
    --> 471         _import(m)
        472         return MODULES[m][0]
        473     elif isinstance(m, dict):
    
    C:\Users\stef\AppData\Local\Continuum\Miniconda3\lib\site-packages\sympy\utilities\lambdify.py in _import(module, reload)
        158 
        159         raise ImportError(
    --> 160             "can't import '%s' with '%s' command" % (module, import_command))
        161 
        162     # Add translated names to namespace
    
    ImportError: can't import 'None' with 'import_module('numexpr')' command

stefsmeets, 2017-09-28:
I noticed this got broken by https://github.com/pycrystem/pycrystem/commit/458972b250021f88f85e7d1101151a3bd03b6013

pc494, 2017-10-13:
I believe this can be worked around as follows:

```
>>> import numpy as np
>>> import pycrystem.diffraction_generator as pcdg`
>>> pcdg.DiffractionSimulation(coordinates = (np.random.random((10,2))),intensities=np.ones(10))
>>> ds.as_signal(10,3,1.0)
<ElectronDiffraction, title: , dimensions: (|10, 10)>
```

stefsmeets, 2017-10-14:
You are missing the assignment to ds in line 3, so it should be:

    ds = pcdg.DiffractionSimulation(coordinates = (np.random.random((10,2))),intensities=np.ones(10))

In any case, it gives me the same error.

pc494, 2017-10-15:
I was unable to reproduce your error on different machine, running the most recent pycrystem.

Currently I think it's to do with _sympy_. Both my suites are running version 1.0, but a little rummage over at hyperspy dug up the commit detailed below:

https://github.com/hyperspy/hyperspy/commit/bf70b5a0c50cfe02e8527547d4f354770fe78ab0

which corrected for the upgrade to _sympy_ 1.1, this commit is on a branch that is not yet released for hyperspy though.

stefsmeets, 2017-10-16:
I was indeed running sympy 1.1.1 with hyperspy 1.3, but if I downgrade to sympy 1.0, I get the same error.

pc494, 2017-10-16:
For now I am out of ideas, hyperspy 1.3 with either sympy 1.0 and sympy 1.1.1 worked fine on my machine.

stefsmeets, 2017-10-16:
No worries, not important for me at the moment, because I can roll back to the old version. I will see if it fixes itself in a future release of hyperspy.

dnjohnstone, 2018-01-08:
So... Do we know the status of this?

pc494, 2018-01-08:
I couldn't replicate the error. In the meantime I've been rewriting the .as_signal() method to remove the hyperspy/sympy dependence for speed purposes, which would fix this.

dnjohnstone, 2018-01-08:
Ok - I'm going to close this then.

closed 2018-01-08 by dnjohnstone

pc494, 2018-01-10:
Fixed by #83