pyxem/pyxem #65: ElectronDiffraction.find_peaks() returning different number of navigation axes

pc494, 2017-11-18:
Having run the following as set up.

```
import numpy as np
import pycrystem as pc
import hyperspy.api as hs

x = np.random.rand(2,2,10,10) # two by two in navigation and 10 by 10 in signal

#generate some maxima 

x[0,0][0][9] = 5
x[1,0][9][0] = 5
x[0,1][9][9] = 5
x[1,1][0][0] = 5

dp = pc.ElectronDiffraction(x)
```

Now moving to a console:

```
>>> dp
<ElectronDiffraction, title: , dimensions: (2, 2|10, 10)>
```

```
>>> dp.find_peaks('skimage')
<DiffractionVectors, title: , dimensions: (2, 2|)>
```

Which is in my opinion is/should be the desired behaviour. However

```
>>> dp.find_peaks('skimage',num_peaks=1)
<DiffractionVectors, title: , dimensions: (2, 1, 2, 2|)>
>>> dp.find_peaks('max')
<DiffractionVectors, title: , dimensions: (2, 1, 2, 2|)>
```

Regardless of what we chose, I think we should try to have .find_peaks() return a consistent object (and write tests to confirm this)

dnjohnstone, 2017-11-18:
The first behaviour is indeed the 'correct' one - good find... We've really got to sort out tests!

pc494, 2017-12-14:
Closed by commit #66

closed 2017-12-14 by pc494