pyxem/pyxem #82: Issue with installing pyxem

M0hsend, 2018-01-09:
We experienced an issue with installing pyxem as it seems to rely on availability of a C compiler. After comparing with pycrystem setup file, we could have it work by modifying the setup.py file, commenting out the lines calling Cython as follows (please kindly advise if there is a more elegant solution to this!):

`#!/usr/bin/env python

import os, sys
from setuptools import setup, Extension

exec(open('pyxem/version.py').read())  # grab version info 

#from Cython.Build import cythonize
#import numpy as np

# if sys.platform == "win32":
    # extensions = [
        # Extension('pyxem.utils.radialprofile', ['src/radialprofile.pyx'], include_dirs=[np.get_include()])
    # ]
# else:
    # extensions = [
        # Extension('pyxem.utils.radialprofile', ['src/radialprofile.pyx'], include_dirs=[np.get_include()])
    # ]
# ext_modules = cythonize(extensions)

setup(
    name='pyxem',
    version=__version__,
    description='An open-source Python library for crystallographic electron'
                'microscopy.',
    author=__author__,
    author_email=__email__,
    license="GPLv3",
    url="https://github.com/pyxem/pyxem",

    classifiers=[
        'Programming Language :: Python :: 3.6',
    ],

  #  ext_modules = ext_modules,

    packages=[
        'pyxem',
        'pyxem.utils',
    ],

    package_data={
        "": ["LICENSE", "readme.rst", "requirements.txt"],
        "pyxem": ["*.py"],
    },
)
`

dnjohnstone, 2018-01-09:
I can't reproduce this issue on the windows machines we have - do you have the error message? Also to rule a whole load of things out, does hyperspy install ok on the same system?

stefsmeets, 2018-01-09:
The thing to do would be to install a C-compiler:
https://wiki.python.org/moin/WindowsCompilers

You can install the **build tools for visual studio 2017** from here: https://www.visualstudio.com/downloads/#build-tools-for-visual-studio-2017

dnjohnstone, 2018-01-10:
Yes - I'm tempted to say that the solution here is to add a note to the dependencies/readme telling people they need to install a C-compiler. The reason I asked the hyperspy question though is that also uses cython and I was curious how they might be making that install without this issue...

bm424, 2018-01-10:
If you're using `conda` I think the solution here is

```bash
> conda install Cython
```

I'll add this to the README now.

M0hsend, 2018-01-10:
Thanks everyone. I am using conda and my hyperspy install is working fine in the same environmnet. 
This is the error I got before attempting any fix:
`(py36_hs) C:\Users\eha56862\Documents\pyxem-master>python setup.py install
Traceback (most recent call last):
  File "setup.py", line 9, in <module>
    from Cython.Build import cythonize
ModuleNotFoundError: No module named 'Cython'
` 
Following @bm424 advice, I installed cython-0.27.3 and now I get the following error:
`(py36_hs) C:\Users\eha56862\Documents\pyxem-master>python setup.py install
Warning: Extension name 'pyxem.utils.radialprofile' does not match fully qualifi
ed name 'radialprofile' of 'src/radialprofile.pyx'
running install
running bdist_egg
running egg_info
writing pyxem.egg-info\PKG-INFO
writing dependency_links to pyxem.egg-info\dependency_links.txt
writing top-level names to pyxem.egg-info\top_level.txt
reading manifest file 'pyxem.egg-info\SOURCES.txt'
writing manifest file 'pyxem.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'pyxem.utils.radialprofile' extension
error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++
Build Tools": http://landinghub.visualstudio.com/visual-cpp-build-tools
`

stefsmeets, 2018-01-10:
`Cython` should be added as a dependency.

stefsmeets, 2018-01-10:
Actually, I just noticed that no requirements are set at all in the setup.py
This class of problems can be avoided by adding all the requirements to the setup command:

`install_requires=['Cython', 'hyperspy', etc...],`

thomasaarholt, 2018-01-10:
With hyperspy it is handled as an optional dependency, as cython is used optionally for vastly speeding up the reading of bcf files. With pyxem it looks like cython is used for fast radial averaging, which I assume is an integral part of pyxem operation, and not an optional feature.

dnjohnstone, 2018-01-31:
We've added install_requires to setup.py and cython to requirements.txt and therefore I will consider this issue resolved. Needing to install a c-compiler is just something people will need to deal with.

closed 2018-01-31 by dnjohnstone