pyxem/pyxem #95: Loading files with pyXem

dnjohnstone, 2018-01-30:
Files loaded with pyXem at present will be in a HyperSpy class e.g. Signal2D class not a pyXem class e.g. ElectronDiffraction. This can be resolved by the following:

dp = pc.load(file)
dp = pc.ElectronDiffraction(dp)

This will be fixed following a more substantial restructuring of the code later this week.

dnjohnstone, 2018-01-31:
Also note that we need to re-write the load method to handle all the pyxem classes properly, but the bonus of the current bug is that anything saved with pyxem will open with the load function which was not the case before.

dnjohnstone, 2018-05-01:
Given #174 -- just to update.

This is now waiting on the hyperspy project to split and create a more generalised "signal register" so that we can resolve this in a suitable way long term.

dnjohnstone, 2018-08-30:
To update on this - this will be handled by https://github.com/hyperspy/hyperspy/issues/1978 and we will then depend on that.

pc494, 2019-05-29:
I think that this can also be resolved for the `hyperspy==1.4`  "fairly" easily

PR #416 (2019-06-06): Improving loading and initializing of class:ElectronDiffraction
---
name: Improving loading and initializing of class:ElectronDiffraction
about: This resolves #95, pxm.load() now gives sensible results (and is easy to extend), ElectronDiffraction can be initialized with a Signal2D and meta-data is maintained.

- [x] ready for review and merge?
---

**Release Notes**
> major
> improvement
Summary:
 - `pxm.load()` will correctly load a saved `pxm.ElectronDiffraction` object
 - `pxm.ElectronDiffraction(s)` will retain metadata for `s` a `Signal2D`

**What does this PR do? Please describe and/or l [... cut]
closed 2019-06-06 by dnjohnstone