pyxem/pyxem #99: Orientation space grids for library generation

dnjohnstone, 2018-01-30:
Symmetry reduced equispaced grids of orientation space, for orientation mapping including all non-equivalent orientations, can be found here for all proper rotational point groups:

https://drive.google.com/open?id=1njQ3AXy3ZDzLaa4gqbN4K5TQULsXuyvW

Eventually we will add utility functions to the code to make the generation of template libraries more user friendly, but in this short term this should help people make progress.

dnjohnstone, 2018-01-31:
For other cases users can build their own grids as follows.

from pyxem.utils.sim_utils import equispaced_so3_grid
grid = equispaced_so3_grid(90,180,90,5)
(and similar for so2) which isn't exactly subtle, but could be massaged for some use cases.

dnjohnstone, 2018-01-31:
More can also be done using this: https://github.com/bm424/texpy - which is where these grids came from.

pc494, 2018-03-07:
Could we confirm the convention used for these?

dnjohnstone, 2018-11-05:
Superseded by #304

closed 2018-11-05 by dnjohnstone