pyxem/pyxem #103: Improving Orientation Mapping

pc494, 2018-01-31:
I've spent some time on this recently, and I think we now have a far more correct implementation.

The next steps are as I see it:

1a) Refactoring the template library in such a way as to store the relevant data for correlate (at the moment `pattern.calibrated_coordinates` and `pattern.intensities`) and prevent these operations being tackled for each and every image.

1b) Removing some of the internal safeguards from correlate, for example:
`in_bounds = np.product((pixel_coordinates > 0)*(pixel_coordinates < shape[0]), axis=1)`
checks that each template peaks lies within bounds of the image, which seems like a good idea. Except that if they don't an error is thrown, which in my opinion is sensible behavior. Reciprocal radius (generating the library) should be such that all peaks land within the image, if this isn't the case it's likely you'll generate poor results as it is.

2x) Making `matching_results` smarter, probably by converting Euler to cubochoric. This would allow us to calculate the distances between various results 'by eye'. texpy make be the saving grace here. 

3experimental) Some improvement on the matching formulae. Either for speed or accuracy. For example you can drop the image normalization if you don't need to do reliability work.

Anyway, updates and thoughts are going to be here going forward. Probably starting with a profiling of the current code.

pc494, 2018-02-01:
For attention of ~ @JMorzy 
Currently` correlate` is ten times slower than it was (due to #98) - fixing this at the moment, hoping to have a WIP pull request up later today.

pc494, 2018-02-01:
#109  is now tracking the changes here.

At time of writing I've completed both (1b) and (3experimental). Point 3 was essential for speed in the end.

pc494, 2018-02-02:
Will close this once I factorize out the normalisation of the template to the library generation step and open a new issue to deal with the wider workflow.

dnjohnstone, 2018-02-02:
Ok a couple of things:

1. I don't think you should do (2) as part of the main code base. It may be good for your workflow and for various reasons, but there is an awful lot of history behind people expecting orientation maps to default output Euler angle based answers. The point is that texpy should make this easy and we should make it easy for the output of orientation mapping to be opened with texpy. Then within a script/notebook people can do this, but it's out of scope in my opinion for pyxem.

2. Don't forget tests.

3. Example notebooks (e.g. including the texpy workflow) would be great.

dnjohnstone, 2018-02-02:
Oh and make sure you find a test case with multiple phases in it! I can give you some data for this next week.

pc494, 2018-02-04:
With regards to (1) - I think I was leaning to a method for the `MatchingResults` class. Seemed most natural. Also how you label your dictionary elements isn't important for execution, so sticking to historical conventions seems like a very good idea.

pc494, 2018-02-06:
Closed by #118

closed 2018-02-06 by pc494