pyxem/pyxem #108: install_requires messing up install at present

dnjohnstone, 2018-02-01:
The specification of install_requires in setup.py is causing an issue at the moment - most easily solved by commenting this out of the setup file but not clear what the underlying issue is.

pc494, 2018-02-01:
Stack trace? I can't reproduce this at present.

dnjohnstone, 2018-02-01:
Hang on.... my hard drive has decided that what I really needed in my life was for it to fail.

JMorzy, 2018-02-01:
The trace is quite long so it's in the file attached. 
[trace_install_bug.txt](https://github.com/pyxem/pyxem/files/1686811/trace_install_bug.txt)
All the issues start with installing dependencies, namely scipy I think.

Hope this helps

pc494, 2018-02-01:
scipy is coming from pymatgen.

Pymatgen moved from wanting scipy 0.14 to scipy 1.0.0 about 2 months ago. So my suggestion is that you install an older version of pymatgen before trying to install pyxem.

At the moment:
`pymatgen==2017.9.23`
is working for me.

Or manually clone at the commit before they bumped the scipy dependency. (go to the git repo, go to setup.py, click blame, CTRL+F for scipy, click to through to the commit, and clone)

dnjohnstone, 2018-03-28:
This seems to have turned out to be a non-issue

closed 2018-03-28 by dnjohnstone