pyxem/pyxem #128: get_radial_profile

tinabe, 2018-03-15:
get_radial_profile in electron_diffraction does not work properly if centers is not constant over the stack. The reason is that a change in the center values could change the length of the average profile, so that the size of the signal axes changes over the stack. This leads to the signal axes not being created at all, and thus the radial profile stack cannot be returned as a signal. 

A proposed solution is to use the center values to calculate the max length and use that for all the average profiles in the stack, by appending zeros where the length is below the max. I have tested it for zero, one and two navigation axes, and it seems to work fine.

tinabe, 2018-03-15:
When trying to push a new branch with the changes upstream, I sadly only get this error: 

> remote: Permission to pyxem/pyxem.git denied to tinabe.
> fatal: unable to access 'https://github.com/pyxem/pyxem.git/': The requested URL returned error: 403

Do you have any suggestions for how to work around that, or for what I could be doing wrong?

pc494, 2018-03-15:
I think we disabled pushing to the main repo? If so you'll need to get the new code into a branch (probably on a fork) and then push your forked repo up to github. You can then submit a pull request on the site. 
The thinking behind this (albeit longer) process is that all new code has to pass the testing suite etc. 

Hope that helps

tinabe, 2018-03-16:
Thank you so much for the quick reply! That makes sense. I tried to do what you proposed, and it seemed to work.

pc494, 2018-03-16:
```
import numpy as np
import pyxem as pxm
from pyxem.utils.expt_utils import radial_average
import hyperspy.api as hs

alpha = np.random.rand(2,2,100,100)
beta = pxm.ElectronDiffraction(alpha)

centers = hs.signals.Signal2D(np.array([[[40,40],[40,50]],[[70,10],[23,23]]]))
centers = centers.transpose(navigation_axes=2)

radial_profiles = beta.map(radial_average, center=centers,inplace=False, cython=False)

#throws an error
rp = beta.get_radial_profile(centers=centers)
```

Would it be agreed that `radial_profiles` is storing the correct information? If so the problem is further on in the method - and I will look into what is going on there.

pc494, 2018-04-01:
After #152 this is no longer a problem in it's current form, instead it requires a bit of (internal) deletions. This issue will serve as a reminder.

closed 2018-04-27 by dnjohnstone