pyxem/pyxem #155: DiffractionVectors for various map() outputs

dnjohnstone, 2018-04-01:
The hyperspy map() function can return a number of data structures now, including ragged arrays. Assigning all of these outputs to a new class e.g. DiffractionVectors is providing difficult without specifying each case individually or getting a brittle result.

Probably surmountable but needs a bit of careful thought and checking hyperspy...

dnjohnstone, 2018-08-01:
DiffractionVectors methods work if:
1. There is a ragged array of vectors i.e. different number per navigation position
2. There is a single nx2 array of vectors

DiffractionVectors methods break if:
1. There are no vectors at any navigation position
2. There are the same number of vectors at every navigation position

This is due to different outputs of map() and is a little complicated to fix - so for now we are noting that they are exceptions and will improve in future.

pc494, 2018-08-03:
DiffractionVectors also breaks if there is a single vector at any given peaks position.

pc494, 2018-08-03:
```
@pytest.mark.parametrize('z, center, calibration, g',[
    (np.array([[100,100],
              [200,200],
              [150,-150]]),
     np.array((127.5, 127.5)),
     0.0039,
     np.array([-0.10725, -0.10725])),
])
def test_peaks_as_gvectors(z, center, calibration, g):
    gc = peaks_as_gvectors(z=z, center=center, calibration=calibration)
    np.testing.assert_almost_equal(gc, g)
```
which is found:

https://github.com/pyxem/pyxem/blob/ba40d7208d622670a48937ea77a2601aee1c6a95/pyxem/utils/expt_utils.py#L387

Looks strange to me. Why is `peak_to_gvectors` taking 3 peaks, and returning only 1 vector?

dnjohnstone, 2019-02-01:
@pc494 where do you think we're at with this since #220 and #255?

pc494, 2019-02-04:
I think it's better, but this issue should remain open as I'm almost certain some thing won't work with non-ragged arrays.

dnjohnstone, 2019-07-15:
@pc494 @shogas do we know of any specific use cases that do not work in the latest version of the code? I think it might be good to close this slightly too broad issue and open more specific issues that could be addressed in manageable PRs...

shogas, 2019-08-01:
I can't think of any specific cases currently, and agree that we should rather open more specific issues if they come up.

It would also be nice if hyperspy provided consistent parameters to the callback sent to the `map()` function, optionally after describing the dimensions in a parameter to `map()`. The function already takes a `ragged` parameter. I have not looked at the implementation in detail. If we find more issues, it could be usefull to compile a list of current workarounds and issues we have in pyxem and start a discussion in the hyperspy repo.

closed 2019-10-04 by dnjohnstone