pyxem/pyxem #166: Removal of functionality

stefsmeets, 2018-04-12:
Hi guys,
I have noticed for the last few weeks that every time I update my copy of pyXem that stuff that I rely on is removed which breaks my code (i.e. <https://github.com/stefsmeets/problematic>). First, I saw that it is no longer possible to obtain the centers of the diffraction patterns (`ed.get_beam_center` if I'm not mistaken), and when I updated today the peak finder I like to use (`ed.find_peaks_regionprops`) is no longer available. What is the reasoning behind all this?

dnjohnstone, 2018-04-12:
Hi Stef,

Sorry that happened! Didn't realise you depended on those functionalities.

Is it just those two things? If so we can just add them back.

The reasons were:

1. We were pushing towards workflows in which the first step is to align the diffraction patterns, which would make getting the beam center unnecessary. On reflection there may be reasons to have non centered centers too so would be happy to add this back.

2. The region props method was removed because we were consolidating methods and improving testing - there were 7 peak finders, that one was not tested or documented and we couldn't see an obvious reason it was better than others -- again adding it back is fine, as now there would be a reason to.

Is there anything else that we removed that you need?

Sorry again - we're really trying to get to a consistent API with full testing etc for some time before summer conferences and in moving quickly towards that this kind of thing might happen. Once we have that platform we'll obey good practice on maintaining API in future.

stefsmeets, 2018-04-12:
Hi Duncan,

No worries, I understand that this can happen. I have a workshop coming up next month where I will be demoing some data processing of serialED data, so these breakages make me a little bit nervous :)

If these functions don't fit with your vision for pyXem, let me now, because I can easily implement them on my side. These two functions are the only ones that were removed.

I guess the `get_beam_center` can be split out of `center_direct_beam`. There are some use cases where having access to the beam center can be useful, for example to characterize the beam position for an experiment (something we check regularly for rotation data). The `find_peaks_regionprops`, is useful to get objects representing each reflection. Just today I was looking at some CBED data, for which I wanted to use this one.

stefsmeets, 2018-04-12:
Let me know if you resolve it. For the mean time I will add this code back to my own library.

dnjohnstone, 2018-04-23:
@pc494 Having thought about this a bit more, I think we should replace both of these functions.

Regarding the centering argument - I agree that's the case for orientation mapping, but there are other experiments where it might make sense to intentionally not put the direct beam in the middle of the detector. In any case I can see situations where you would want to know how the direct beam has moved, even if you centre the patterns in the end anyway - so I don't see much harm in it. It would be fine to just document clearly that orientation mapping needs centering first.

pc494, 2018-04-23:
Seems reasonable. Having them (centring and finding centres) as separate methods of `ElectronDiffraction` seems most natural to me. ie ~

`dp.center_direct_beam() #inplace centering` 
`dp.get_direct_beam_center() #not in place`

dnjohnstone, 2018-04-27:
Ok - I've added back the get_direct_beam_center() method and will do the peak finder at some point today.

closed 2018-07-24 by dnjohnstone