pyxem/pyxem #178: Centering when the direct beam isn't the brightest beam

pc494, 2018-05-02:
Current methods put the brightest beam in the center. Given the reasonableness of the physical assumption (the direct beam will generally lie near the center of the image square) I think we could automated the "draw a square around the direct beam" idea.

smc204, 2018-05-02:
It seems plausible, although it may be worth checking how far the direct beam translates relative to the length of possible g-vectors. If the translation of the direct beam is larger than the size of a g-vector this might still not work, but having it as an option seems eminently sensible. 

Are you thinking of this functioning as a crop or as a mask? For things like FFT methods (e.g. subpixel centering method) it will probably be important to retain the original pixel dimensions as a mask. I could see that reducing the number of pixels (cropping) would speed things up in other peak finding/centering methods, though.

pc494, 2018-07-23:
This is going to be implemented in a code (centering) tidy up some time in late July/early August

pc494, 2018-08-07:
#228 closed this.

closed 2018-08-07 by pc494