pyxem/pyxem #182: Opening crystallographic map in MTEX

fuk21, 2018-06-20:
A crystallographic map produced by the save_map function in pyXem creates a horizontally flipped image when imported as an EBSD file in MTEX. I'm guessing this is because in pyXem the origin point is at the top left corner, while in MTEX it's at bottom left. A possible solution is to convert all the y-indices to (max y-index - original y-index), but I'm not sure how easy/difficult it'd be to put this into the code.

pc494, 2018-06-20:
Will investigate.

pc494, 2018-06-20:
By "horizontally flipped" do you mean through 90 degrees? If so, I think this is to do with ordering of arrays in python vs matlab (C vs F) [1].

Messing around with 
`plotx2east`
(and similar for z) in MTEX seems a suitable workaround.

I've limited enthusiasm to change this unless there is a really good reason.

[1] - see for example ~ https://bic-berkeley.github.io/psych-214-fall-2016/index_reshape.html

fuk21, 2018-06-20:
No, flipped as in it’s upside down.

pc494, 2018-06-21:
I'm pretty sure this is better dealt with in MTEX. 
pyXem will save with x=0 being the leftmost column and y=0 being the the upper row.

closed 2018-06-21 by pc494