pyxem/pyxem #183: PyPI packaging?

magnunor, 2018-06-26:
It would be nice having pyxem packaged in PyPI, as currently (I think) it is only available pre-packaged via conda-forge?

pc494, 2018-06-28:
My personal feeling is that this (a PyPI package) is best left until we a 1.0 version and the willingness to commit to semantic versioning. Despite the recent slow up, a lot of development (and API incompatible changes) still need to be made between now and 1.0

magnunor, 2018-06-28:
I think having 0.y.z in PyPI is fine, since the 0 major version implies that the API can break at any point:

>  Major version zero (0.y.z) is for initial development. Anything may change at any time. The public API should not be considered stable.

(From https://semver.org/)

Also, many python packages on PyPI do this: dask, skimage, HyperSpy, and all of my libraries.

dnjohnstone, 2018-08-02:
To be addressed in #208

closed 2018-08-02 by dnjohnstone