pyxem/pyxem #184: Common data processing "backend" library?

magnunor, 2018-06-26:
Currently there are several python based data processing libraries for working with pixelated STEM/4D STEM/precession, which all have some kind of overlap in data processing functionalities.

For example, radial integration is implemented in both pixStem, fpd and pyXem:

- https://gitlab.com/pixstem/pixstem/blob/master/pixstem/pixelated_stem_class.py#L457
- https://gitlab.com/fpdpy/fpd/blob/master/fpd/fpd_processing.py#L696
- https://github.com/pyxem/pyxem/blob/7bdac8d78593c5b056b3f6803bf24245faf319b6/pyxem/signals/electron_diffraction.py#L375

And likewise with removal of dead pixels.

Would there be any interest in combine these efforts into one "backend" library, which handles these types of more generic processing like radial integration, centre of mass, dead pixels, ...

Ideally it would only depend on the more common scientific python stack, so potentially other non-HyperSpy based projects could also use them. I'm tempted to also include `dask` in this, since it has proven _very_ useful in working with the large pixelated STEM datasets I've got.

dnjohnstone, 2018-07-24:
This is probably a good idea - I'm not likely to write anything before September at this point, but I'd be supportive of this happening and you're welcome to just rip anything out of pyxem that's helpful. Would be particularly good if this results in everything being better tested.

dnjohnstone, 2018-08-01:
This is not really a pyxem issue so closing here.

closed 2018-08-01 by dnjohnstone