pyxem/pyxem #187: Carrying kwargs over to map in our methods

pc494, 2018-07-19:
Take for example:

```
    def apply_affine_transformation(self,
                                    D,
                                    order=3,
                                    inplace=True,
                                    *args, **kwargs):
        """Correct geometric distortion by applying an affine transformation.
        Parameters
        ----------
        D : array
            3x3 np.array specifying the affine transform to be applied.
        inplace : bool
            If True (default), this signal is overwritten. Otherwise, returns a
            new signal.
        Returns
        -------
            ElectronDiffraction Signal containing the affine Transformed
            diffraction patterns.
        """
        return self.map(affine_transformation,
                        matrix=D,
                        order=order,inplace=inplace)
```

Now, things like `show_progressbar` won't be carried over to map, as kwargs aren't handled here. Likely to be other methods like this. Will run a sweep as and when.

dnjohnstone, 2018-09-11:
@pc494 is this now dealt with? It's certainly improved in many places, but I haven't checked everything systematically, have you?

pc494, 2018-09-11:
I have not. My thoughts on this are:
1) It was only really an issue for the specific progress bar issue we were having, which has a more elegant solution
2) It will be fixed in totality when the testing is made more structured
3) Given these we can probably close this issue in the form it currently exists.

closed 2018-09-11 by pc494