pyxem/pyxem #208: RELEASE 0.6

dnjohnstone, 2018-08-01:
A checklist of miscellaneous actions that should be done for a tidy 0.6 release:

- [ ] Website needs an overhaul given the various changes to functionality and in particular given the removal of functionality that is documented there.
- [ ] Tests for scalable_reference_pattern.py
- [ ] Tests for tensor_field.py
- [ ] Tests for diffraction_component.py
- [ ] Remove all cython code
- [ ] Document Signal/Generator/Component code structure
- [ ] Documentation bibliography
- [ ] Open WIP PRs for re-addition of various functionalities on current workplan

pc494, 2018-08-01:
cython code is done in #207 
PyPi packaging I'm happy to do, but it's obviously the very last thing on this list
Happy to overhaul large chunks of the website too

pc494, 2018-08-04:
Think it might be worth adding **dependencies checking** to this. Just added (6 months) of pymatgen code to my old install (because we changed pyxem to reflect the changes they had put in) and it's now broken somewhere else.

pc494, 2018-09-03:
The steps to this are now pretty ordered:

1) Changes to the API
- [ ]  Diffpy (#240)
Followed by:
- [ ]  Vector based strain mapping
- [ ]  Vector based orientation mapping

2) Non API changes
- [ ] Clean up the testing suite to help remove old/weak bits of code
- [ ] Improve the docstrings
- [ ] Overhaul the website
- [ ] Packaging (PyPi and Conda #243)

pc494, 2018-09-09:
I think we should also pick a logo at this point. I'll play around with a colour scheme on the hyperspy hypercube unless anyone has any better ideas...

pc494, 2019-01-31:
The check list at the top looks good to me.

closed 2019-04-23 by dnjohnstone