pyxem/pyxem #244: Diffraction library

juliesnilsen, 2018-09-03:
Yesterday I updated pyxem for the first time since April, and now the get_diffraction_library function seems to have changed.  I get the error: 'dict' object has no attribute 'struct_lib', and was wondering if this is a bug or am I defining my structure library in the wrong way? My code was working fine until the update. For now I will just reinstall the old version of pyxem, but I hope that someone can answer :)

dnjohnstone, 2018-09-04:
Thanks @juliesnilsen, do let us know if that doesn't work for you.

I've flagged this as a 'documentation bug' so that we remember to update the documentation. We're aiming for a much more stable release of the code some time in the next month or so!

dnjohnstone, 2018-11-05:
superseded by #304

closed 2018-11-05 by dnjohnstone