pyxem/pyxem #250: Currently not making full use of Hyperspy 1.4

pc494, 2018-09-07:
It seems likely that most user will want to update to 1.4 for (among other things) logarithmic plotting, however it also has subpixel alignment written in (from a really old pull request). If we make use of this for 0.7 we may have to move ourselves to 1.4=< only. 

- [ ] In the short term I'll update the requirements to suggests 1.4, as this should remove some of the warnings from the testing (https://github.com/hyperspy/hyperspy/pull/2025)

dnjohnstone, 2018-09-08:
Ok 2 things:

1) I don't think we need to do anything about sub-pixel align2D()? That's inherited and a user can use it if they want in workflow. To my mind it's sufficiently different in methodology to the center_direct_beam() method that it doesn't matter if we have both?

2) Looking ahead - I'd expect that within the next year we become hyperspy 2.0 =< only (because splitting of the package will have a big impact on us). So I don't think we may as well just keep shifting with the times on that.

dnjohnstone, 2018-09-08:
Logarithmic plotting is however something we can use better for sure.

pc494, 2018-09-09:
This is all good. My thinking on (1) was merely if the hyperspy version proves far better we may as well wrap it, but that's more a potential enhancement than an issue, and when/if it gets done we can adjust accordingly. (2) is obviously very true, so I'm going to close this.

closed 2018-09-09 by pc494