pyxem/pyxem #280: Is affine transformation supposed to change dtype?

hwagit, 2018-10-10:
When performing an affine transformation on diffraction patterns of `uint8` data type using `pyxem.signals.electron_diffraction.ElectronDiffraction.apply_affine_transformation`, scikit-image's [warp function](http://scikit-image.org/docs/dev/api/skimage.transform.html#skimage.transform.warp) changes the pattern's data type to `float64`. Is this the intended behaviour of `apply_affine_transformation`?

Passing `preserve_range=True` into `apply_affine_transformation` does not change this behaviour, meaning that it is not getting passed to `warp`?

An example is included below.

```python
>>> import pyxem as pxm
>>> import numpy as np
>>> dp = pxm.load('data.hspy')
>>> dp = pxm.ElectronDiffraction(dp)
>>> dp.data.dtype
dtype('uint8')
>>> dp.apply_affine_transformation(np.array([[1.025, 0, 0], [0, 1.065, 0], [0, 0, 1]])
HBox(children=(IntProgress(value=0), HTML(value='')))

>>> dp.data.dtype
dtype('float64')
```

To preserve file size, I just reverted the data type using `dp.change_dtype('uint8)`. However, I guess it would be better if `apply_affine_transformation` did this? Or `warp`?

I'm using

```bash
pyxem                     0.6                       <pip>
scikit-image              0.14.0           py36hfc679d8_1    conda-forge
```

pc494, 2018-10-10:
This is a bug (relates to #187) - thank you for raising it. Should be an easy fix because - as you say - it just involves getting kwargs passed to the correct functions. I will fix it in the next couple of days.

dnjohnstone, 2018-10-10:
I don't think this is a bug actually... and I think the **kwargs are being passed through to warp in this case (although that is an issue in other parts of the code). The behaviour is justified as follows:

skimage.transform.warp() converts things to double precision in order to avoid issues with interpolation when applying the transformation.

The preserve_range argument doesn't do anything about the dtype it's just that if you don't pass it the range goes to -1 to 1 on conversion to float whereas if you do pass it it is rescaled to 0 to 255.

All of this is, in my opinion the correct behaviour, and what currently happens in pyxem.

Even if you rescale to uint8 you want the operation done as float and I think it's fine to leave that to the user - you'll need the RAM anyway to do the rescaling.

pc494, 2018-10-11:
Duncan is correct about what preserve range is doing. The functionality one actually needs is described at https://scikit-image.org/docs/dev/user_guide/data_types.html 

I agree that type conversion can be left to the user.

closed 2018-10-11 by pc494

hwagit, 2018-10-11:
Ah, I misunderstood `preserve_range`, thanks for clearing that up!