pyxem/pyxem #286: DiffractionLibraryGenerator::get_diffraction_library matrix convention

shogas, 2018-10-12:
**Describe the bug**
I think the `DiffractionLibraryGenerator::get_diffraction_library` function passes the wrong matrix (transposed) to diffpy Lattice contructor for `baserot`. This might have happened during the transition from pymatgen to diffpy.

**To Reproduce**
See [shogas/sped_processing_playground/pyxem_rotation_bug.ipynb](https://github.com/shogas/sped_processing_playground/blob/master/pyxem_rotation_bug.ipynb).

**Expected behavior**
Rotating around the first z axis should not change the pattern intensity or relative spot positions, only rotate the pattern. (I think at least. It could be my understanding that is wrong).

**Desktop (please complete the following information):**
Not at all relavant, but:
 - OS: Windows
 - Browser Firefox
 - Version 79ec2175dea1119d032afd5487a89e28e4aa86c6

shogas, 2018-10-13:
Thank you for clearifying.

For the documentation: I think my misunderstanding is that the rotation list should rotate the "camera" coordinate system, not the atomic structure? Those are "oposite" operations. That would explain why I got my matrices inverted (which explains why, when changing the lattice, diffpy changes the atomic positions with the inverse matrix, keeping the atoms at the same absolute position). I don't think this is because of extrinsic vs intrinsic coordinate systems, but the cases could be equivilant.

pc494, 2018-10-15:
@shogas I'll get back to you when I write the documentation for this. 

For the time being I think that you are correct. pyxem rotates the crystal rather than the lab axes, see [1], primarily for the clarity of the internal code. However in this case the conversion to standard Bunge axes seems to be simply a multiplication by -1.

[1] https://www.ctcms.nist.gov/~langer/oof2man/RegisteredClass-Bunge.html)

dnjohnstone, 2018-10-15:
@pc494 I'm not sure what the code is currently doing, but I do think that we want to be specifying the orientations in the standard Bunge convention you reference above - mostly because we want our "crystallographic map" answers to be in Bunge convention for later analysis?

dnjohnstone, 2018-10-15:
Also - is this something to document on the website? It seems a bit strange for a docstring thing?

pc494, 2018-10-15:
In answer to the latter, yes, it's a website thing IMO - it's just not done yet (https://github.com/pyxem/pyxem/blob/master/docs/source/conventions.rst)

And to the former, internally we rotate the reciprocal lattice rather than the Ewald sphere. This for me (with my developer hat on) seems like the best way to do it, but I think an internal factor of -1 might be sufficient to mean that "externally" we're in the agreed upon convention and internally we still work in the way we would like to. This is how we deal with degrees (external) and radians (internal) for example.

dnjohnstone, 2018-10-17:
This is basically just a bit of a convention minefield - as illustrated/discussed here: http://iopscience.iop.org/article/10.1088/0965-0393/23/8/083501

I think it's also changed several times in the lifetime of pyxem as 3 of us that have touched that bit of code were thinking about it from different perspectives.

The convention that I personally tend to default to is "Orientations are conventionally defined as passive rotations (i.e. tensor quantities are not rotated) that transfer coordinates with respect to a crystal reference system into coordinates with respect to a specimen reference system." This is, for example, inverse to the convention that was used in the book by Bunge and also the link provided above. (Sorry - I made a slip in my previous comment when I forgot that Bunge did things that way). So in this case I think that, if the comments from @pc494 are correct - we're currently in the convention that I tend to use and I believe this is also the standard convention used by MTEX...

So.... basically we just need to finish that page of the website...

dnjohnstone, 2018-10-17:
For the website - I guess we want:

(1) A description of the convention we're using for orthonormal to crystallographic axis alignment
(2) A description of the convention we're using for Euler angles / orientations
(3) Some examples of how that rotates basic crystal structures.
(4) A warning about the convention minefield that directs people to that paper by Rowenhorst et al...

pc494, 2018-10-17:
This sounds good. We'll have to test to find the answer to (1) - for (2) I think we should try to do the same as MTEX.

dnjohnstone, 2019-01-30:
Is this now resolved?

closed 2019-02-01 by dnjohnstone