pyxem/pyxem #287: Improving the peak finder documentation - specifically for the 'xc' method

pc494, 2018-10-16:
**Describe the bug**
This is a documentation issue with the xc peak finder. Because of where users access it from (`dp.find_peaks`) it's very difficult to work out how to use it. This can be fixed now, although (DISCUSSION POINT) splitting the peak finder methods would go a long way to solving this problem.

dnjohnstone, 2018-10-17:
I'm a bit reluctant to split out lots of different peak finders - just think it will be a bit messy but you're obviously correct that there is a documentation challenge. I think there are a few things:

(1) We probably want fewer peak finders in an ideal world - we've already had an issue once with removing some though, so this is a bit of a problem. On the sooner we do it the fewer problems there will be in the long term.

(2) We should probably suggest people use find_peaks_interactive() before running any full peak finding operation --> that has 2 advantages, (1) it makes people think about the values they're using, and (2) the interactive notebook GUI makes the **kwargs/*args evident.

(3) The documentation doesn't have to all be in docstrings - I think a lot of this could be solved by improving the website (http://pyxem.github.io/pyxem/basics.html#peak-finding) and then referring to that in the docstring.

pc494, 2018-10-23:
(1) If we really care about this the obvious place to start is by cutting one of the matrix methods. I think they are no longer the fastest option (xc is quick) - which suggests keeping the "slower better" one (LoG). No longer feel quite so strongly about this though.

(2) #296 (+docs in #297) makes this an even better suggestion

(3) And an improvement to the website generally (and one that suggests (2)) I think this problem is more or less a solved one.

closed 2018-10-23 by pc494