pyxem/pyxem #291: pep8 compliance slacking

dnjohnstone, 2018-10-20:
The code should comply with the pep8 style guide: https://www.python.org/dev/peps/pep-0008/

This is starting to get a bit lax in parts of the code. We should correct this - probably with the help of autopep8: https://pypi.org/project/autopep8/

We also need to ensure that contributing documentation is clear that contributors need to work to this style guide.

pc494, 2018-10-25:
I'm going to brute force this (with autopep8) before 0.6 (probably after #293 is merged) but I think it probably makes sense to follow the recommendation of [1] in future, and I'll add something to the contributing guide to that effect.

[1] https://stackoverflow.com/questions/14328406/tool-to-convert-python-code-to-be-pep8-compliant

pc494, 2018-10-29:
Closed by #306

closed 2018-10-29 by pc494