pyxem/pyxem #327: Abstract ElectronDiffraction and subclass Map/TiltSeries

dnjohnstone, 2019-01-19:
At the moment most of pyxem is based around the idea that data was acquired in a scanning electron diffraction mapping experiment where a diffraction pattern is recorded at every probe position in a scan. Many of the methods could, in principle, be applied to electron diffraction patterns recorded in a tilt series but we'll need to improve the abstraction of the appropriate bits of code.

As a first step I propose subclassing ElectronDiffraction to an ElectronDiffractionMap class and splitting map specific functions appropriately. At some point in the future an ElectronDiffractionTiltSeries.

This will wait for #315 to be merged to avoid lots of merge conflicts.

dnjohnstone, 2019-02-01:
@pc494 do we still agree that we want this before 0.7 release? I think we do...

pc494, 2019-02-04:
70-30 I think yes we do want to do this before 0.7

dnjohnstone, 2019-07-16:
This has been deprecated by #430

closed 2019-07-16 by dnjohnstone