pyxem/pyxem #329: Suggestion: StructureLibrary design

shogas, 2019-01-23:
`StructureLibrary` currently inherits from `dict`, but does not use it. Instead it contains a `struct_lib` member, which leads to code like `structure_library.struct_lib` several places in the code. Should either the inheritance be removed for clarity, or the functionality of `struct_lib` be moved directly into the class itself, so that you can do

    structure_library = StructureLibrary(...)
    structure_library['phase name']

instead of

    structure_library = StructureLibrary(...)
    structure_library.struct_lib['phase name']

dnjohnstone, 2019-01-23:
Thanks for raising this issue - I also find this behaviour odd and it does need to be changed one way or the other. I think both your suggestions are potentially good solutions.

The current structure comes from @pc494 - and I remember chatting to him about why this was this way, but I don't remember what the answer was - so perhaps he can write it here?

pc494, 2019-01-23:
I (mildly) favoured a layout that allowed things like:

`structure_library.orientations`

Rather than having the figure out what the phase were called and then access them via the dictionary. I agree that `structure_library.struct_lib` is a little ugly though. My preference would be to remove the (obviously incorrect) inheritance from `dict` at the `class` level.

dnjohnstone, 2019-02-01:
Did we settle on the best solution here? I think we should change this before the 0.7 release. I think I'd favour just removing the dict inheritance and leaving it there for now - do we agree on that?

pc494, 2019-02-01:
Yes @dnjohnstone, I agree, the `dict` inheritance is dumb but the current code structure isn't too bad.

dnjohnstone, 2019-02-01:
addressed by #345

closed 2019-02-01 by dnjohnstone