pyxem/pyxem #347: Crystal map orientation result display

shogas, 2019-02-06:
The `CrystallographicMap.get_orientation_map()` function currently converts the euler angles representing the orientation to an axis-angle pairs using `euler2axangle` and displays the angle (in radians) after discarding the axis. This might not be very informative except for showing changes across the sample.

Can we find another single-value measure to display the orientation map? Angle from the z-axis is one option? A better approach would be to use a color map as described in "Orientations - perfectly colored" by Nolze and Hielscher and implemented in MTEX, but we might want a simpler solution for now.

(Partially in reply to an email exchange with @dnjohnstone)

pc494, 2019-02-06:
I feel that anything more complicated than the `get_distance_from_modal_angle` work flow should be dealt with by MTEX / texpy in the near/medium term

dnjohnstone, 2019-02-06:
Yeah I think here converting to axis-angle and plotting the angle is ok (although it should be in degrees) as the goal is just for the user to get a picture that lets them see the rough grain structure in a polycrystalline sample and believe that their result is not totally ridiculous. I.e. check there are some grains and it's not just random noise.

For anything more advanced we have a save function to give something that can be analyzed in MTEX and doing that would be the recommendation.

shogas, 2019-02-06:
Then the current implementation is correct. I'll change it to degrees and create a PR along with a couple other orientation-related fixes later today.

pc494, 2019-02-07:
Closed by #349

closed 2019-02-07 by pc494