pyxem/pyxem #350: Improving error checking for _"xyz"_get_experimental_square

pc494, 2019-02-18:
At the moment if you have a size zero square (due to an upstream bug) you get the following

```
  File "/home/pc494/miniconda3/envs/miniconda-1/lib/python3.6/site-packages/pyxem/generators/subpixelrefinement_generator.py", line 154, in _com_experimental_square
    z_adpt[0,:] = 0

IndexError: index 0 is out of bounds for axis 0 with size 0
```
Adding a bug catch somewhere could save users (me) a lot of trouble. I'll do this in the near future.

dnjohnstone, 2019-04-21:
@pc494 please could you give a timeline for this one?

pc494, 2019-04-21:
Before play opens on Tuesday, got delayed with #369 yesterday annoyingly.

dnjohnstone, 2019-04-21:
Great, thanks - it was mostly a question about whether it's a v0.7 or v0.8 thing - will aim to have v0.7 out on Wednesday so I'll put it with that.

dnjohnstone, 2019-04-21:
Resolved by #374

closed 2019-04-21 by dnjohnstone