pyxem/pyxem #353: TestXCmethods.test_peaks_xc fails on macOS

tschoonj, 2019-03-14:
**Describe the bug**
I am currently [submitting](https://github.com/conda-forge/staged-recipes/pull/7980) a recipe for pyxem to conda-forge but the TestXCmethods.test_peaks_xc is failing on macOS, both on Travis-CI and Azure...

Links to the failures: [Travis-CI](url
https://travis-ci.org/conda-forge/staged-recipes/builds/506163298?utm_source=github_status&utm_medium=notification) and [Azure](url
https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_build/results?buildId=21517&view=logs&jobId=c572e5d5-7ba7-55ce-5575-7ce1d9f7161e)

**To Reproduce**
Run the testsuite on macOS

**Expected behavior**
Test succeeds

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here. eg) The use of an uncommon system/version

shogas, 2019-03-14:
Reproduced on windows. Some more details:

```
_________________________ TestXCmethods.test_peaks_xc _________________________

self = <test_peakfinders2D.TestXCmethods object at 0x0000023C0F64D438>
peaks = array([[50, 80],
       [41, 41],
       [41, 40],
       [40, 41]], dtype=int64)

    @pytest.mark.filterwarnings('ignore::FutureWarning')  # skimage not us
    def test_peaks_xc(self, peaks):
        disc = np.zeros((4, 4))
        disc[1:3, 1:3] = 1
        peaks = find_peaks_xc(peaks, disc, 3)
>       assert peaks.shape == (2, 2)
E       assert (4, 2) == (2, 2)
E         At index 0 diff: 4 != 2
E         Use -v to get the full diff

tests\test_utils\test_peakfinders2D.py:103: AssertionError
```

tschoonj, 2019-03-14:
Interestingly the test runs fine on the windows AppVeyor and Azure in my conda-forge PR...

Also the macOS assert fails in a slightly different way:

```
>       assert peaks.shape == (2, 2)
E       assert (3, 2) == (2, 2)
E         At index 0 diff: 3 != 2
E         Use -v to get the full diff
```

shogas, 2019-03-14:
The test mostly calls two skimage functions. My version (from conda list) is:
`scikit-image              0.14.0           py36h6538335_1    conda-forge`

pc494, 2019-03-14:
I think (although hard to know without testing in Windows/Mac) that the peakfinder is returning different results on different OS systems. It seems unlikely that this is to do with the OS systems specifically. As a first port of call if you are not already makes sure you're install with the 
`-r requirements.txt` flag set (`pip install . -r requirements.txt`). Note that this will give you `scikit-image 0.14.0` so it may well be another package further down the tree.

dnjohnstone, 2019-03-14:
@tschoonj Thanks very much for this and for making a conda-forge recipe... That's been something on our todo list for a long time so very much appreciated! Will have a look in a bit - incidentally i'm currently at Diamond delivering a hyperspy workshop today and tomorrow - I'd have some time later this afternoon or tomorrow morning if having a quick chat about pyxem might be useful.

tschoonj, 2019-03-14:
ah cool, I am in the big office next-door 😄 

I am leaving in half an hour but tomorrow should be fine.

shogas, 2019-03-16:
Some more updates (fresh install of pyxem in new conda environment, Windows 10):

The test fills two areas, a 3x3-area and a 2x2-area and tries to cross correlate with a 2x2 disc. On my setup, the `match_template` function creates a 1x1-peak at [y, x] = [51, 81], and a 2x2 peak at [y, x] = [41:43, 41:43].

`peak_local_max` returns four points, one for the single 1x1 peak, and three for the 2x2-peak. Following the documentation, I would actually have expected 5 peaks (1 for the 1x1, 4 for the 2x2), since ([skimage peak_local_max docs](http://scikit-image.org/docs/dev/api/skimage.feature.html#skimage.feature.peak_local_max))

> If there are multiple local maxima with identical pixel intensities inside the region defined with min_distance, the coordinates of all such pixels are returned.

Where is this behaviour differing from a working run? I can quickly set up a linux instance if this is hard to debug by messaging.

dnjohnstone, 2019-04-18:
This was resolved by #365

closed 2019-04-18 by dnjohnstone