pyxem/pyxem #354: `pxm.ElectronDiffraction.find_peaks()` on 1D navigation axes fails

jordiferrero, 2019-03-21:
**Describe the bug**
When you run the `pxm.ElectronDiffraction.find_peaks()` function on a 1D navigation axis `ElectronDiffraction` object: e.g. (x | x*, y*), it fails.

It is trying to find a second navigation axis, when assigning:

- `x = self.axes_manager.navigation_axis[1]` : runs fine
- `y = self.axes_manager.navigation_axis[1]` : fails

**Expected behavior**
This function should wwork on both 1D (x | x*, y*) and 2D (x ,y | x*, y*) navigation axes, given an 2D signal.

pc494, 2019-03-22:
This has been known about for a while. Consensus was (I believe) won't fix, it's an edge case and would be quite involved to hack through with hyperspy. I've tended to just make <x 2 | x* y*> signals and gone from there.

dnjohnstone, 2019-03-22:
Thanks @jordiferrero, as @pc494 says this is something that we know about and it actually applies to many things in the code, it's a little bit complicated because it's linked to the fact that we get "ragged arrays" back from thinks like peak finding, i.e. a different number of peaks in each pattern, and these can behave in slightly unfriendly ways. Ultimately I think we clearly do need to fix this behaviour, but in the short term it's kind of low on the priority list....

ericpre, 2019-07-23:
It looks to me that this should be fixed in https://github.com/hyperspy/hyperspy/pull/1497, @jordiferrero could you checkout that branch and confirm?
You may need to checkout one of the gui branches too, links to PRs are in https://github.com/hyperspy/hyperspy/pull/1497#issuecomment-495654049.

dnjohnstone, 2019-07-24:
I think the bug in pyxem is to do with the assignment of the peak finding results to the DiffractionVectors class rather than a problem with the peak finders. The hyperspy PR looks good to me and it's probably not a bad idea to upstream this (and maybe other code) to there at some point.

We're going to overhaul all of the functionality associated with diffraction vectors when we do pyxem-0.10.0 though and that's when we'll deal with this properly from a pyxem perspective.

ericpre, 2019-07-24:
Ok, it sounds like there are potentially several bugs, since I fixed a few bugs with ragged arrays in the hyperspy/hyperspy#1497 (and the `find_peaks` code in pyxem is very similar to the one of that PR).
@jordiferrero, if you can provide a traceback and a minimal example, it would help to understand what is/are the bug(s) and to fix the issue properly.

dnjohnstone, 2019-07-25:
Ok - yeah, I'm sure you improved the code in hyperspy/hyperspy#1497 - do you think we could try to get that in to hyperspy-1.5.0 and then we can work on fixing any remaining bugs on this side after that?

I'm sure myself and @pc494 could spend a little time looking at that PR if that's all we need - it does look basically there to me...

dnjohnstone, 2020-07-31:
Resolved by  hyperspy/hyperspy#1497

closed 2020-07-31 by dnjohnstone