pyxem/pyxem #357: SubpixelrefinementGenerator interface

shogas, 2019-03-26:
Currently, if I read the code correctly, the subpixel refinement refines a list of P peaks on each of the diffraction images in an NxM ElectronDiffraction instance, creating NxMxP refined peaks. What is the use case for this?

I would think that the refinement is a natural step after peak finding,

```
peaks = dp.find_peaks(...)
refgen = SubpixelRefinementGenerator(dp, peaks)
peaks_refined = refgen.center_of_mass_method(...)  # Or conventional_xc
```

where `peaks` is a DiffractionVectors object with a list of peaks at each navigation position and `peaks_refined` is the same, but with refined positions. This is possible with the current implementation, but requires more work for the user and requires some knowledge of the internals of the signal classes.

I have a change ready that supports this, but I wanted to ask if the current interface has a use case I am missing (or what I want is already supported) before I put any work into polising and creating a pull request.

pc494, 2019-03-27:
Simon,

I agree with you, it would seem natural to turn a list of peaks into a list of refined peaks, however I opted against this initially for a couple of reasons.

1) Not having ragged peaks made writing the code a lot easier, and I'm lazy.
2) Not having different peaks at each navigation position also makes life simpler.
3) Subpixel refinement is quite slow, so minimising the amount that is done seemed like a good idea.
4) The only use case we had at the time (and still perhaps?) is strain mapping, which requires peaks to be in (more or less) the same place, so putting a small numpy array at something like the unstrained position seemed clean (and avoided requiring users playing around with the signal class in this use case).

If you don't mind me asking what is your use case (happy to receive as an email if not suitable for public publishing)?

shogas, 2019-03-27:
Thank you. Yes, strain mapping is a good fit for the current interface.

I am looking at improvements for the vector matching, and the algorithm seems to be quite sensitive to the peak finding step. I probably don't really need much more than pixel accuracy, but the current peak finding algorithms often miss by one or two pixels, which is significant for 144x144px signals. This is somewhat improved by enough parameter fiddling, but any of the peak finding algorithms (even just finding local maxima without further processing) combined with some refinement seems to give better results with almost no user input.

Speed will probably be a problem. I will keep a fast path for matching a single peak list across all navigation positions if I end up changing the implementation.

Feel free to close the issue if there are no further comments.

pc494, 2019-03-27:
I think it's worth leaving open, as a guide to a potential future direction.

dnjohnstone, 2019-03-27:
@shogas your suggestion sounds good to me. thanks! We'll close this after your new implementation has been merged via PR just so that we have this conversation as a record until then.