pyxem/pyxem #381: Add pyXem news feed

dnjohnstone, 2019-04-25:
It would be good to add a news feed to the website to document various ways pyxem engages with people.

dnjohnstone, 2019-04-26:
To remind us of some news items to include:

- presenting at 2018 ePSIC hyperspy workshop 
- presenting at 2019 ePSIC hyperspy workshop 
- electron diffraction workshop at NTNU (2019) https://www.ntnu.edu/web/temgemini/diffraction2019
- hyperspy workshop at miroscopy and microanalysis (2019) https://www.microscopy.org/MandM/2019/program/short_courses.cfm
- european crystallographic meeting software fayre (2019) http://www.cryst.chem.uu.nl/lutz/thursday.html

pc494, 2019-05-08:
I've now moved this to https://pyxem.github.io/pyxem-website and will make the adjustments needed here to streamline the code.

pc494, 2019-05-08:
Once #396 is merged this will be done and can be closed. 

I think @dnjohnstone will just needs to change the URL in the repo header, and take the website offline in the settings page (of this repo). The "new" one is a little rough around the edges, but having pulled it out we can now make changes pretty quickly.

closed 2019-05-09 by dnjohnstone