pyxem/pyxem #383: Factor out simulation focused code to standalone package

dnjohnstone, 2019-04-26:


pc494, 2019-04-27:
This closes #57

dnjohnstone, 2019-05-16:
A plan for this based on discussion around splitting hyperspy is:

- [ ] create pyxem/diffsims
- [ ] in v0.9.x, move/copy what is needed (tests, utilities functions) to a pyxem/pyxem/diffsims/ folder.
- [ ] run git filter-branch for the pyxem/pyxem/diffsims folder as explained in https://help.github.com/articles/splitting-a-subfolder-out-into-a-new-repository/
- [ ] push the new local repository to pyxem/pyxem/diffsims
- [ ] update repository structure, add setup.py, licence, etc.
- [ ] setup CI (keep appveyor and travis?)
- [ ] release diffsims as a separate package
- [ ] update v0.9.x to depend on the now separate diffsims package

dnjohnstone, 2019-07-15:
The existing simulation focused code has now been separated out at: www.github.com/pyxem/diffsims

There are now two steps:

1. Finalize and release diffsims package v0.1.0
2. Create a new version of pyxem, probably v0.10.0, that depends on diffsims and removes all of the current simulation code.

dnjohnstone, 2019-07-16:
This is deal with in #429

closed 2019-07-18 by pc494