pyxem/pyxem #384: Update requirements

dnjohnstone, 2019-04-26:
requirements should be updated.

pc494, 2019-04-29:
I have actioned this in #387, the conclusion of which is as follows:

- There isn't a whole load of point us having a "standard" requirements file, and quite a few others don't (pandas, numpy) so I'm going to remove that.

- It does make sense to have some stable environments, so that code that may fail due to version changes can be tested against a working environment 

- Because it's conda based these need to be different for different environments (linux/OSX/Windows, Py3.7 Py3.6) it'll be a little time consuming to make them all.

What I propose is making a 3.7 Linux environment for `v0.8` and then completing the set (ideally for  3.7,Linux,Windows,OSX for `v0.9`).

If the above all seems reasonable I'll shuffle the github tiles to confirm that and get the `v0.8 `implementation wrapped and completed.

dnjohnstone, 2019-05-03:
Ok, I think this does sound like a sensible plan although also not a particular priority I don't think, so up to you whether you want to try and get it done for 0.8

pc494, 2019-05-09:
The full version (ie beyond #397) of this will be a bit more complicated, and will be left until `0.9.0`

dnjohnstone, 2019-07-15:
@pc494 what was left to be done here?

pc494, 2019-07-15:
I had grand plans, but looking back over it I think the best bet is to mark this as `wontfix` and close the issue. Packages will update, this will break the build, we will always have to go fix it, but we're getting better at that. I'll leave you @dnjohnstone to close.

closed 2019-07-15 by dnjohnstone