pyxem/pyxem #423: Local gaussian method handling of bad peaks

shogas, 2019-06-17:
**Is your feature request related to a problem? Please describe.**
Currently, when the maximum in `expt_circle` for a vector in the `local_gaussian_method` refinement is located on an edge, an error is raised: https://github.com/pyxem/pyxem/blob/81ff463c6a3445fcf7ab7fa8132540600537a90c/pyxem/generators/subpixelrefinement_generator.py#L251

If a single peak in a single diffraction pattern hits this condition, no refinement is done in the entire signal.

**Describe the solution you'd like / alternatives**
One of:
 - Return a shift of `(0, 0)`. If the maximum is on the edge, the refinement is missing information to create a good fit, and should leave the peak as is.
 - Return a shift to move the vector to the maximum.
 - Remove the vector from the list of peaks.
 - Find the maximum only inside the square, leaving a border as padding.
 - Leave the implementation as is, and require the user to filter the peaks first.
 - Other options I have not thought about.

Either of these can be combined with issuing a warning.

pc494, 2019-06-17:
I've been using this functionality quite a lot recently, and I actually think this clunky behaviour is pretty good. A pixel maximum at the edge almost certainly means your square size is wrong, and forcing users to correct this seems like a good idea; especially as if one choses to issue a warning you're probably going to get 1,000's of them due to the `map`.

This also points to a second issue, which is that I don't currently use the `DiffractionVectors` version, perhaps we could have different code pathways for that vs `numpy.ndarray`?

pc494, 2019-07-12:
I've flipped on this, I think return a shift `(0,0)` is better. Will also raise a single warning (to avoid the `map `1000 warning problem)

dnjohnstone, 2019-07-18:
Resolved by #438

closed 2019-07-18 by dnjohnstone