pyxem/pyxem #426: Deadline Approaching for 0.9.0

pc494, 2019-07-12:
So I've checked the dates on this pending release, M&M starts August 4th, ECM August 18th. On the assumption that 0.10.0 will (only) address #407 what do people want/expect to have in for 0.9.0 - I for example hope to have:

- Lattice fitting (#425) and some minor fixes to the subpixelrefinement generator + working with tensors (#338)
- Some housekeeping on the loading function so that it ships basically finished  

That does however leave quite a lot of stuff still open in the Milestone. Thoughts?

dnjohnstone, 2019-07-17:
I've now marked all existing issues and PRs with milestones consistent with this:

v0.9.0 - various changes that we want in before M&M

v0.10.0 - to be released almost immediately after v0.9.0, and before M&M, just removing simulation code to diffsims

v0.11.0 - everything else that is going to be on a longer timeframe

dnjohnstone, 2019-07-17:
@tinabe if you'd like to get your bit of code for determining unique vectors by clustering into the code the 0.9.0 release would be a good opportunity to do that.

dnjohnstone, 2019-07-18:
These points are now all addressed and v0.9.0 will follow all open issues / PRs being resolved

closed 2019-07-18 by dnjohnstone

tinabe, 2019-07-18:
> 
> 
> @tinabe if you'd like to get your bit of code for determining unique vectors by clustering into the code the 0.9.0 release would be a good opportunity to do that.

Yes, that would be good. I'll put it in for you to review this week. What is the actual DL for this version in terms of dates, Aug 4th?

pc494, 2019-07-18:
> 
> Yes, that would be good. I'll put it in for you to review this week. What is the actual DL for this version in terms of dates, Aug 4th?

Sadly it's a fair bit earlier than that as course materials for the M&M workshop need to be in (@dnjohnstone can confirm exact data) - perhaps middle of the next week?

dnjohnstone, 2019-07-18:
It'll need to be fully merged before 23rd July. No big deal if not, will just have to wait for the next release