pyxem/pyxem #428: How to use pyXem to index arbitrary rotations

tcpekin, 2019-07-15:
Hello,

I would love to use pyXem to map the phase and orientation of randomly oriented crystalline samples, with diffraction patterns acquired in transmission (similar to ACOM). Is this possible? The worked example only does in plane rotations, and I must admit I haven't been able to figure out via the documentation how to (successfully) extend it to arbitrary rotations, so I've started with just limiting it to in plane rotations. I believe the settings I'm using are roughly correct to generate the library, as I can generate a similar looking diffraction pattern compared to my data using the calculate_ed_data method, but when I go to index and plot to see if it has worked, I get nonsense. All of the red markers are clustered in the top left corner, at a very different scale than the diffraction pattern.

[image]

Zoomed into the top left pixel:
[image]

Any help would be appreciated, thanks so much,

Thomas Pekin
HU Berlin

pc494, 2019-07-15:
Treating this as two questions:

1) Yes you can do out of plane rotations, I'll see if I can dig something out to make it obvious how to do that.

2) You'll need to calibrate your signal to get sensible plotting of markers. `ElectronDiffraction` has a `.set_diffraction_calibration()` method that should do the trick (obviously being careful with units!)

pc494, 2019-07-15:
Further to (1) ~ you've got two main options, the path involving the generator is @shogas's code, I'll see if he's got a better demo notebook kicking around. Meanwhile if you are comfortable generating a rotation list in euler angles "by hand/mtex" you can use

`lib = StructureLibrary(identifiers=['name of structure'], \\`       
                                   ` structures=[diffpystructure], \\ `
                                   ` rotation_list[[(a,b,g),(a1,b1,g1),(a2,b2,g2)]])`

where a(lpha), b(eta), and g(amma) are angles in degrees in the Bunge convention (passive rotations)

tcpekin, 2019-07-16:
Thank you very much @pc494, I think this has really helped. At least the markers are at the correct order of magnitude now, and I have a way of naively (the results look like latitude/longitude, so not equally spaced, but using Euler angles) generating angles that I believe are appropriate, to cover the half sphere for a monoclinic system. At a < 10 degree separation in XZX, I run into a memory error, in that my system cannot perform the correlate step, but I guess that just is a natural limitation based on how I choose my orientations. Thanks so much for the immediate help.

dnjohnstone, 2019-07-16:
Hi @tcpekin - thanks very much for raising an issue on this. Great that you're using pyxem and if you're at all interested in becoming a contributor/develop, please do let us know, we're always keen to expand the base!

You can get a list of appropriate Euler angles using the rotation_list_stereographic function here, it may be similar to what you're doing.

https://github.com/pyxem/diffsims/blob/master/diffsims/utils/sim_utils.py

As you might notice with that we're currently refactoring all of the "simulation" code out of the pyxem repository for various code efficiency reasons, but the user interface from pyxem should not change when we do that.

Somewhere I'm pretty sure I have a jupyter notebook demonstrating this - I'll try to dig it out in the next week ish.

If we can help more, just ask.

tcpekin, 2019-07-18:
Hi @dnjohnstone - thanks for all the help, that is exactly what I'm looking for. With regards to expanding the user base, I/we are very interested on using some open source Python code (almost certainly pyXem) to replace the ACOM system we currently have here at Humboldt, so there might be a few more questions in the future. The other questions I have are more general, with regards to preprocessing the data and other topics, in the future would you prefer a github issue per question? Thanks again for all the help.

pc494, 2019-08-07:
@tcpekin - I think my preference would be stackoverflow questions for use cases (like part 2 of this issue) - it seems like a natural home for this sort of thing (compared to gitter/googlegroups). You might want to raise github issues linking the question for the first few so that we can get used to it (so that a `pyxem` tag can be created and followed). Obviously there is likely to be some crossover, as some "user issues" will be bugs, but other projects (notably `tensorflow`) have worked around this.

All that said, if you put problems somewhere obvious one of us will likely find them!

dnjohnstone, 2019-09-17:
@tcpekin did you figure this out, or do you still need some help here?

tcpekin, 2019-09-17:
Hi @dnjohnstone, I kind of put this project on the back burner, but now since you've mentioned this, I've taken another look at it. 

I think I mentioned this at M&M, but since one of the structures is monoclinic, when I'm building my library of diffraction patterns over all of the rotational space, I run out of memory. So I've done a less highly sampled library through the various Euler angles, but the indexing when plotted aren't fantastic. I think it is a combination of poor data quality (I have tried some preprocessing of the diffraction patterns, but it is unclear if it is enough), not fine enough sampling of the template library, lack of a large number of diffraction spots in some patterns, and then, finally, if my experimental parameters are correct (which I think they are, as my `template_library.diffraction_generator.calculate_ed_data(structure_library.structures[0], 10, False)` looks qualitatively right). 

So, my current plan is to move onto some radial averaging and then classifying using these vectors, and to return to pyXem once we have some slightly higher quality data. I would love to replace ASTAR in our workflow with pyXem, but I think this initial dataset is just a little too hard to learn with. 

Actually, with that in mind, do you have public polycrystalline datasets which are indexed using pyXem? I like the in plane rotation while on zone axis example, but it doesn't have enough complexity for me to fully understand the parameter space I have to work with when using pyXem. 

Thanks for bringing this back up,
-Tom

dnjohnstone, 2019-09-18:
Ok, I think I'll do 3 things to tie this off in the short term:

1. We'll raise an issue about being more memory efficient with template libraries as that's clearly an issue but will require a fair bit of coding to fix.

2. For indexing based on radially averaged data - I also went a little way down that avenue a couple of years ago, but never really finished it. Here (https://github.com/pyxem/pyxem/blob/master/pyxem/generators/indexation_generator.py) you'll find a ProfileIndexationGenerator, which should assign hkl values to a load of vector lengths. This is not really documented because it's not really finished, but if you're interested in contributing this would be an awesome place to start and I'd be happy to make that bit of the code work.

3. I'll have a think about a shareable polycrystalline dataset - the best would be the stuff from @shogas masters thesis - perhaps he can tell us where all that ended up as I've forgotten.

shogas, 2019-09-18:
I'd certainly be OK with sharing/creating some examples from my thesis, but I'd have to ask Ton for which (parts of the) datasets can be shared. I'm guessing one of the gold NP datasets is closest to a polycrystalline sample.

tcpekin, 2019-09-20:
Sorry for the slow response, but that sounds like a great plan to me. I will check out 2) shortly. With regards to the polycrystalline dataset, gold NP datasets sound great, as that seems relatively close to my dataset, of a randomly oriented polycrystalline sample with small grain size. Thanks for all of the assistance.

closed 2019-11-09 by dnjohnstone