pyxem/pyxem #431: pyxem does not install in winpython using pip

ericpre, 2019-07-17:
Because of the version constraint for matplotlib (and scikit-image for the released version):
https://github.com/pyxem/pyxem/blob/c36da7bbb346298771519d97b44aed2c87582092/setup.py#L50-L51
See appveyor log, line 337
https://ci.appveyor.com/project/ericpre/hyperspy-bundle/builds/26038725#L337

This is using winpython 2019-02rc

dnjohnstone, 2019-07-17:
Thanks @ericpre for the report - we're going to release v0.9.0 by the end of this week and will do something about this at that time.

dnjohnstone, 2019-07-17:
Resolved, I think, by #437

dnjohnstone, 2019-07-17:
Seems we have an appveyor issue now - it's a slightly odd looking one, and didn't cause an issue in PR but is in the main repo now.... Reopening as this was the last thing merged...

dnjohnstone, 2019-07-18:
This now appears to be an unrelated AppVeyor issue.

closed 2019-07-18 by dnjohnstone