pyxem/pyxem #445: Add plot of dp processed with range of DoG background subtractions somewhere

dnjohnstone, 2019-07-29:
@shogas had this neat little code snippet in a notebook for making a plot where the navigation axes are sigma_min and sigma_max of a DoG background remover as a way of manually identifying good parameters. Seems handy so we should put it somewhere (I took it out of the notebook because it was a bit cluttering, but not unhelpful)

dp_test_area = dp.inav[0, 0] 

gauss_stddev_maxs = np.arange(2, 12, 0.2)  # min, max, step
gauss_stddev_mins = np.arange(1,  4, 0.2)  # min, max, step
gauss_processed = np.empty((
    len(gauss_stddev_maxs),
    len(gauss_stddev_mins),
    *dp.axes_manager.signal_shape))

for i, gauss_stddev_max in enumerate(tqdm(gauss_stddev_maxs, leave=False)):
    for j, gauss_stddev_min in enumerate(gauss_stddev_mins):
        gauss_processed[i, j] = dp_test_area.remove_background('gaussian_difference',
                                                          sigma_min=gauss_stddev_min, sigma_max=gauss_stddev_max,
                                                          show_progressbar=False)
dp_gaussian = pxm.ElectronDiffraction2D(gauss_processed)
dp_gaussian.metadata.General.title = 'Gaussian preprocessed'
dp_gaussian.axes_manager.navigation_axes[0].name = r'$\sigma_{\mathrm{min}}$'
dp_gaussian.axes_manager.navigation_axes[0].offset = gauss_stddev_mins[0]
dp_gaussian.axes_manager.navigation_axes[0].scale = gauss_stddev_mins[1] - gauss_stddev_mins[0]
dp_gaussian.axes_manager.navigation_axes[0].units = ''
dp_gaussian.axes_manager.navigation_axes[1].name = r'$\sigma_{\mathrm{max}}$'
dp_gaussian.axes_manager.navigation_axes[1].offset = gauss_stddev_maxs[0]
dp_gaussian.axes_manager.navigation_axes[1].scale = gauss_stddev_maxs[1] - gauss_stddev_maxs[0]
dp_gaussian.axes_manager.navigation_axes[1].units = ''

pc494, 2019-07-30:
Will add this to my friday list (if nobody else fancies it)

pc494, 2019-08-07:
Was closed by #451

closed 2019-08-07 by pc494