pyxem/pyxem #447: Add pattern match optimiser

dnjohnstone, 2019-07-29:
@shogas produced a pattern match optimizer (https://github.com/shogas/sped_processing_playground/blob/master/template_param_optimize.ipynb), which seems pretty neat - we should put it in the demos or somewhere in the code if we can find a suitable location for it.

dnjohnstone, 2019-09-17:
@stefsmeets do you think this would still be useful, given #468 is now in?

stefsmeets, 2019-09-17:
This is an optimizer for the pattern matching, which is different altogether. I think it could be useful, because from my understanding the orientations from pattern matching are quite discrete (only at the simulated angles). An optimizer could help with increasing the precision and hopefuly accuracy of the orientations.

closed 2023-01-16 by pc494