pyxem/pyxem #448: Add filter_peaks method to DiffractionVectors class

dnjohnstone, 2019-07-29:
@shogas had the following code snippet in a notebook that would be a sensible and useful addition to the DiffractionVectors class - we should add it.

def filter_peaks(peaks):
    peaks = peaks[0]
    # Only keep vectors within max length and remove centre (closer than 5px to image centre)
    return peaks[(np.linalg.norm(peaks, axis=1) < reciprocal_radius_max) &
                 (np.any(np.abs(peaks) > 5 * reciprocal_angstrom_per_pixel, axis=1))]

peaks.map(filter_peaks)

pc494, 2019-07-30:
I wrote something similar for @tinabe quite recently, so I'm happy to tidy this up and get it in.

dnjohnstone, 2019-10-04:
@pc494 do you think you could wrap this one up for 0.10.0?

pc494, 2019-10-04:
Yep can do.

dnjohnstone, 2019-11-07:
There may be something to do still here - but we now have a method that does this.

closed 2019-11-07 by dnjohnstone