pyxem/pyxem #456: Use metadata to track processing of datasets

dnjohnstone, 2019-08-17:
Great idea from @stefsmeets to track processing steps using metadata, which can stay with a saved file by creating a new node for "Processing"

I.e.

class serialED(Signal2D):

    _props_collection = []
    _centers = []
    _raw_orientations = []
    _orientations = []

    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.metadata.add_node("Processing")
        self.load_extras()

See here: https://github.com/stefsmeets/problematic/blob/master/problematic/serialED.py

dnjohnstone, 2019-09-19:
@pc494 your opinion on whether we should track processing would be appreciated. In particular how we might associate "results" classes with underlying raw data.

dnjohnstone, 2019-09-19:
A list of things that we might want under various headings:

Instrument:
- Type
- Beam energy
- Camera length

Detector:
- Type
- Pixel size

Processing:
- Raw data
- Processing parameters

pc494, 2019-09-19:
While I certainly agree we don't do a great job with metadata at the moment, I'm not that enthusiastic to track processing with it right now. That said I think keeping the location of the file from which a (processed) data set comes from is a good idea.

dnjohnstone, 2019-09-19:
> While I certainly agree we don't do a great job with metadata at the moment, I'm not that enthusiastic to track processing with it right now. That said I think keeping the location of the file from which a (processed) data set comes from is a good idea.

Could you expand a bit on why?

pc494, 2019-09-19:
So it falls into two main reasons, (1) a lot of the code still changes pretty rapidly, so if your method use was a couple of months ago it's likely you would get (slightly) different results if you ran them differently. (2) Not having metadata forces users to have clearer workflows. 

It must be said these aren't hugely strongly held opinions on my part.

dnjohnstone, 2019-09-19:
> So it falls into two main reasons, (1) a lot of the code still changes pretty rapidly, so if your method use was a couple of months ago it's likely you would get (slightly) different results if you [...]

Ok, so my thinking is:

(1) storing "version" in processing would solve this? And actually be a good thing for understanding why what you might think was the same processing, actually wasn't.

(2) The fact is that sometimes you have to come back to something after multiple years and metadata will definitely still be there, whereas even a very good processing notebook is more likely not to be. The same would hold for sharing partially processed datasets.

shogas, 2019-09-19:
I like the idea of storing information about the processing if enough is included to reproduce the results. The "processing pipeline" I used for my thesis stored the name of the raw data file, the date, version (git sha) of the code and all input parameters in a separate metadata file. I found that useful to be able to repeat earlier runs.

Versioning should really include the full conda environment to be fully reproducable, but that is probably not practical. Just the pyxem version helps a lot if we assume that the environment is a mostly separate install with just pyxem and dependencies.

Processing steps that generate new objects might require some thought to ensure that all parameters are included. For example, for a correlation result from template matching, you probably want the parameters used to contruct the structure library and diffraction library as well.

The example metadata above allows multiple processing steps, but it should possibly also include the order in which they were executed. Python dicts are guaranteed to be in insertion order as of 3.7, and we could rely on that if it works through serialisation/deserialisation. The structure also allows users to add metadata of their own processing steps to the same structure, which is good.

pc494, 2019-09-23:
> Processing steps that generate new objects might require some thought to ensure that all parameters are included. For example, for a correlation result from template matching, you probably want the [...]

We could use a `list` of `dicts`? `dicts` have been unordered for so long I would far prefer appending to a `list`, and it could be created at `__init__` time with the first element as a `dict` that contains not only all the import metadata, but the version of pyxem as well? Then each subsequent method runs an `.append()` statement.

dnjohnstone, 2019-10-12:
This is going to be best addresses when XRD is also considered and will therefore be pushed to 0.11.0

dnjohnstone, 2020-01-13:
Another assorted list of parameters we may wish to record:

- radiation type (X-ray/electron)
- beam energy
- camera length
- detector pixel size (real space units / reciprocal space units)
- scan step size (real space units)
- detector size (pixels)
- instrument_id
- rocking angle/frequency/type
- distortion calibration

We might also want to be able to work in various units, or track which units we are working in:

- k = 1/A
- q = 2pi/A
- s = 4pi/A
- 2theta (degrees, less favorable as radiation dependent)

dnjohnstone, 2020-02-23:
@chrisallen80 and @M0hsend - we spoke offline about defining key metadata for scanning diffraction / 4D-STEM experiments.

This would be a good place to write down any key parameters that you feel would be good to track.

dnjohnstone, 2020-03-21:
@ericpre - I wondered if you might have any opinion / insight on how we should best be making use of metadata in a hyperspy-esque way?

CSSFrancis, 2020-04-02:
One thing we discussed at the developers meeting the other day was including additional N-D signals which were the same size as the navigation signal. I will share my thoughts on the matter and then we can go from there. 

I think adding a node to the metadata called Navigation_signals would be the best place to start.  Each navigation signal would have the same axes as signal.axes_manager.navigation_axes and could be flagged as use_as_navigation_plot = True in line with PixSTEM's ability to use a predefined signal for visualization.  It might be worth thinking about having multiple navigation signals plotted i.e. you wanted to visualize thickness and some dark field reconstructed image at the same time.

The signals that I can think of right now are:

- HAADF
- Sample Thickness Map
- Probe Current per Pixel
- Virtual Dark Field
- Virtual Bright Field

There are probably many more options to consider, which suggests that there might be an argument for the user to add any signal they want to as long as the axes match the navigation axes.

Another thing to consider is operating on these signals.  Two examples of this in my own work 
- In many of my experiments we can't capture the zero beam so we use the HAADF image to measure the thickness at every position. 
- When doing FEM experiments thickness filtering is important because direct comparisons only hold for samples of similar thicknesses. 

In one case we are using one of the navigation signals to create another navigation signal and the other we are using a navigation signal in a function. 

I think this argues for a set of predefined methods to generate singals that can be used in functions and a generic one to save whatever the user desires.  Maybe a flag in the virtual dark field generator to save_as_navigation_signal. 

Just some basic examples of how these functions and the metadata might work

`diffraction.add_nav_signal(data, name, unit=None, use_as_navigation_plot=False)  # generic case
diffraction.add_HAADF(data, unit=None, use_as_navigation_plot=False)
diffraction.add_thickness(data, unit=None, use_as_navigation_plot=False)`

```
├── Navigation_signals

       ├──HAADF
       |    ├──data
       |    ├── use_as_navigation_plot=False
       |    ├── unit = None
       ├──Thickness
       |    ├──data
       |    ├── use_as_navigation_plot=False
       |    ├── unit = "nm"
       ├──MySignal
       |    ├──data
       |    ├── use_as_navigation_plot=True
       |    ├── unit = None
```

pc494, 2023-01-19:
closing as stale (nobody is working on this as of now).

closed 2023-01-19 by pc494