pyxem/pyxem #466: v1.51 of Hyperspy. ValueError: Need at least one array to stack

norskie, 2019-09-02:
How to reproduce:
Installing pyxem according to [https://pyxem.github.io/pyxem-website/](url) using Miniconda leads to a bug related to v1.51 of Hyperspy, which seem to be traced all the way back to Numpy. 

One way to reproduce the bug is to run the example notebook 
[https://github.com/pyxem/pyxem-demos/blob/master/01%20GaAs%20Nanowire%20-%20Data%20Inspection%20-%20Preprocessing%20-%20Unsupervised%20Machine%20Learning.ipynb](url)

where the traceback is:
[https://imgur.com/a/2cwwyoT](url)
(error message_ 

I have used both Ubuntu and Windows - both with fresh installs of pyxem/miniconda - and the problem is the same.

Reverting to hyperspy v1.41 seems to fix the problem, however. 

The issue is adressed at Hyperspy's github:
[https://github.com/hyperspy/hyperspy/issues/2236](url)

pc494, 2019-09-02:
Thanks for this, I agree with your assessment, although I think it's the numpy version that should be pinned until a bugfix is released for hyperspy.

I'm tempted to make a bugfix release for pyxem 0.9 which pins `numpy==1.16` so that the default conda install etc gives the correct answer (@dnjohnstone would you agree?)

dnjohnstone, 2019-09-02:
Yep a 0.9.1 would be good - there are actually a few things in the master (fixing load_mib for example) that were probably bugfixes and therefore shouldn't be in master.... So maybe even a 0.10.0 would be good, or just do a 0.9.1 off master?

pc494, 2019-09-02:
I will build `0.10.0` off the master I think, it'll have some new (not backwards compatible) strain map things + the first bits of Stef's work.

shogas, 2019-09-02:
Seems like you already have a solution for this, but in the meantime, for anyone who has already installed pyxem, a temporary fix is to manually revert either numpy or hyperspy (which also changes the numpy version).

If you use the Anaconda GUI, you can search for the numpy package, and there should be a button to change the numpy version.

If you use the command line, activate the correct environment with `conda activate <name of your environment>`, then re-install numpy to a lower version using `conda install numpy=1.16.4`. You can see which version you have of a package by using `conda list`.

dnjohnstone, 2019-09-02:
> I will build `0.10.0` off the master I think, it'll have some new (not backwards compatible) strain map things + the first bits of Stef's work.

Ok - I've merged up two further PRs one from Stef and the numpy one. Assuming there's no oddness with testing of that merging it would be great if you could do a 0.10.0 release!

dnjohnstone, 2019-09-02:
Although we are only 10 days before our planned 0.10.0 release so I personally could be persuaded to wait until that date and try to tidy up our TODO by then. We really should have done a 0.9.1 but that's easy to say with hindsight... Maybe we just accept we'll have a 0.11.0 in 10 days time?

pc494, 2019-09-03:
> Although we are only 10 days before our planned 0.10.0 release so I personally could be persuaded to wait until that date and try to tidy up our TODO by then. We really should have done a 0.9.1 but that's easy to say with hindsight... Maybe we just accept we'll have a 0.11.0 in 10 days time?

`0.10.0` was originally earmarked for 10th October (5 weeks away) - on the flip side it's not that hard for me to backport and make a `0.9.1`. Let me know.

norskie, 2019-09-05:
Thank you for the quick replies, glad I could help identify the problem. 
Until the problem is fixed, a work-about for Linux(using Conda) is: 

1. Start with a fresh environment
```
$ conda create -n pyxem
$ conda activate pyxem
```

2. Install Numpy v1.6.4 https://anaconda.org/anaconda/numpy
```
$ conda install -c anaconda numpy 
```

3. Install Pyxem 
```
$ conda install -c conda-forge pyxem
```

Also, the problem should be fixed in https://github.com/hyperspy/hyperspy/pull/2247.

If you agree I could edit the original post with the work-about and perhaps close the issue?

dnjohnstone, 2019-09-05:
Thanks @norskie - let's just leave this here as it is - I think people should be able to find it here. Hopefully there will be a 1.5.2 release of hyperspy soon and this problem will be resolved. We can close the issue after that.

dnjohnstone, 2019-09-17:
Now resolved by hyperspy-1.5.2

closed 2019-09-17 by dnjohnstone