pyxem/pyxem #477: VDFImage is an inconsistent class name

dnjohnstone, 2019-09-17:
Classes in pyxem generally build on the Signal1D/Signal2D type naming structure, except for VDFImage.

VDFImage is also a bit misleading since it's really used for general plots of diffracted intensity which may or may not be dark-field images.

DiffractedIntensity2D is a possible alternative, but am aware that this is not really established terminology in any way.

dnjohnstone, 2019-10-10:
Do we have any opinions on what this name should be? @pc494 I'd like to make this change for 0.10.0 so we get lots of API changes out of the way in one go...

pc494, 2019-10-10:
`DiffractedIntensity2D` is fine by me to be honest.

pc494, 2020-07-31:
This was dealt with by #565 as far as I can tell, closing.

closed 2020-07-31 by pc494