pyxem/pyxem #479: Loading .mib files and axis flipping

dnjohnstone, 2019-09-17:
In #391 we added an axis flip to loading .mib files. In #472 we got rid of that (I think). We should either decide which is more useful or make this an option

M0hsend, 2019-09-17:
@dnjohnstone See here: Adding flip option to load_mib #481

dnjohnstone, 2019-09-17:
Resolved by #481

dnjohnstone, 2020-03-07:
Somehow this issue has resurfaced - well actually the how is obvious, there are no appropriate unit tests for load_mib, which is a problem.

Anyway, not sure how/why we seem to have reverted the change introduced in #481 

@pc494 or @M0hsend - do you know why that happened?

dnjohnstone, 2020-03-07:
Anyway, what this really highlights is that we need someone to actually finish this file loader properly and write unit tests for it so that this kind of thing doesn't happen.

The knock on impact is massive - I've just had to tell some people that they need to reprocess around 250 SED datasets because this has been unstable for so long.

M0hsend, 2020-03-09:
Sorry to see this happen @dnjohnstone . I don't recall submitting a separate PR to potentially reversing this option so not sure how this happened. Totally agree that we need proper tests for this.

dnjohnstone, 2020-03-09:
> Sorry to see this happen @dnjohnstone . I don't recall submitting a separate PR to potentially reversing this option so not sure how this happened. Totally agree that we need proper tests for this.

Yeah - I don't think it was you, it was quite probably me!

It might have been because the flip possibly required a .compute() so it wasn't lazy any more and this then defeated the point. But I really can't remember.

dnjohnstone, 2020-03-25:
Resolved by #571

closed 2020-03-25 by dnjohnstone