pyxem/pyxem #483: Differentiating qualtiatively different types of diffraction patterns

dnjohnstone, 2019-09-18:
The issue of naming diffraction-related classes precisely came up in #409 

There are qualitatively different types of diffraction pattern, which require different kinds of treatment. The most obvious example is that when you record a convergent beam electron diffraction CBED pattern the size of the diffraction disks can vary widely, overlap, not overlap, or be quite small.

Terminology in recent literature in this area is a bit of a mess, as everyone seems to want to call everything nano, but I think the historically established terminology is:

KosselMoellenstedtPattern - for a diffraction pattern with non overlapping disks (small convergence angle)

KosselPattern - for a diffraction pattern with overlapping disks (large convergence angle)

PrecessionPattern - for a diffraction pattern recorded using a double conical rocking geometry, which typically also has non-overlapping disks, although this is not a necessary condition.

At present, nearly everything in pyxem is implemented for the KosselMollenstedt and/or precession type situation.

Of course there are also EBSD patterns that are clearly ElectronDiffraction too.

dnjohnstone, 2019-09-18:
Options could be:

1. Creating different classes for all these different cases.

2. Storing some information about what we have in metadata/attributes.

3. Simply documenting what our scope is more explicitly.

pc494, 2019-09-18:
I'm strongly in favour of the 3rd of these until further notice.

dnjohnstone, 2019-09-18:
Yeah, I'm not particularly inclined to change this either - but I thought I may as well write this thought down so it's clear we've at least considered it.

dnjohnstone, 2019-10-04:
I'm going to close this as on consideration I think we're happy with our current naming structure.

closed 2019-10-04 by dnjohnstone