pyxem/pyxem #509: Error when running test suite using pytest --pyargs

ericpre, 2019-10-12:
See for example https://ci.appveyor.com/project/ericpre/hyperspy-bundle/builds/28049557.

dnjohnstone, 2019-10-12:
Do you know what that error means? It doesn't like how we've named something?

dnjohnstone, 2019-10-12:
Also, is that using the previous 0.9.2 release? Just trying to figure out what's changed because I thought we were testing with pyargs at some point...

ericpre, 2019-10-13:
It was using what was available on pypi at the time of the build (11/10/2019).

I have not had a look at it but it seems that the test suite is not setup to run out of the source directory.

dnjohnstone, 2019-10-15:
The main thing I don't understand is that I think we run pytest --pyargs pyxem in our .travis.yml

And travis is doing just fine on continuous integration...

ericpre, 2019-10-17:
I don't know, it could be platform dependent or winpython specific. One way to check would be to run from the winpython distribution. Anaconda type distribution seems to work fine including on windows: https://dev.azure.com/ericpre/hyperspy-bundle/_build/results?buildId=208

dnjohnstone, 2019-11-09:
@pc494 do you understand this at all? I can only conclude that this a WinPython specific thing right now - but I don't really have any idea what's actually causing it....

pc494, 2019-11-09:
I do not know what is going with this. It seems very odd. I will mention however that all of the errors are thrown on tests that no longer existed in the `0.9.2`, testing diffsims functionality etc, so I would be curious to know if this still apllies for `0.10.0`

dnjohnstone, 2020-03-23:
@ericpre Do you know if this is still an issue with the latest version of pyxem?

ericpre, 2020-03-23:
Yes, it still seems to be the case:
https://ci.appveyor.com/project/ericpre/hyperspy-bundle/builds/31655850

ericpre, 2020-03-24:
On the master branch (not the release version), it seems to be working fine now:
https://ci.appveyor.com/project/ericpre/hyperspy-bundle/builds/31687698

closed 2020-03-24 by ericpre