pyxem/pyxem #514: ValueError in chunked_application_of_UDF

mvonlany, 2019-10-31:
"ValueError: need at least one array to concatenate..." in using chunked_application_of_UDF.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to chunked_application_of_UDF, with do_centering applied (defined below*)
2. Click on 
3. Scroll down to
4. See error [shown in screenshots, unsure whether the a shared version of the notebook is saved]

**Expected behavior**
Return centred diffraction pattern

**Screenshots**

[image]

[image]
[image]

**Additional context**
Add any other context about the problem here. eg) The use of an uncommon system/version 

*def do_centering(dp):
    dp.center_direct_beam(method='cross_correlate',
                                 radius_start=4,
                                 radius_finish=8,
                                 square_width=25)
    return dp

pc494, 2019-10-31:
Thanks for raising this issue. I'll have a further look tomorrow after a provisional scan didn't turn much up.

In the meantime could I ask for a clarification? It appears you are running a user defined big data function that doesn't reduce the size of the data. If you can't load all the patterns into memory, when the `pyxem` functionality attempts to do that (with the centred patterns) things are likely to go wrong. I'm not convinced that is the case here but I wanted to check what we had wasn't a use case issue.

pc494, 2019-11-01:
The following MWE reproduces the problem described in the issue.

```
import pyxem as pxm
import hyperspy.api as hs
import numpy as np

#make data
s = np.random.rand(16,16,64,64)
s = hs.signals.Signal2D(s)
s.save('temp')

def do_centering(dp):
    dp.center_direct_beam(method='cross_correlate',
                          radius_start=4,
                          radius_finish=8,
                          square_width=25)
    return dp

from pyxem.utils.big_data_utils import chunked_application_of_UDF

x_list = [0,5]
y_list = [0,5]
fp = 'temp.hspy'

chunked_application_of_UDF(fp,x_list,y_list,do_centering)
```

I think it's an off by one error in the `for` loop at `line 71`. Hoping to fix it today, although a little concerning that the tests didn't pick it up.

dnjohnstone, 2019-11-05:
@pc494 did this get fixed somewhere?

PR #516 (2019-11-08): Bufix for user defined functionality workflow
---
name: Bufix for user defined functionality workflow
about: Fixes #514 & should work out the box (@mvonlany) and can be merged straight away, thus making it in for `0.10.0` - just.

---

**Release Notes**
> bugfix
> Summary: (n/a, this feature is new in `0.10.0`)
closed 2019-11-08 by dnjohnstone