pyxem/pyxem #528: Possible bug in get_orientation_map()

EirikOpheim, 2019-11-14:
I suspect that the attribute get_orientation_map() of the class CrystallographicMap interprets the euler angles as radians, while they should be interpreted as degrees. I have done template matching for simulated data, with the list of orientations in my simulated data as my template library. I have used Au particles, with a_0 = 4.08

Steps to reproduce the behavior:
1. Simulate data at different orientations. I used a grid of angles ranging from (0,0,0) to (90,45,0) with 5 degree increments along the x and y axes.
2. Use list of orientations used in simulation to generate template
3. Perform the following sequence:
   - indexer = IndexationGenerator(test_data, template_library)
   - match_results = indexer.correlate(n_largest=4)
   - cryst_map = match_results.get_crystallographic_map()
   - ori_map = cryst_map.get_orientation_map()
   - ori_map.plot(cmap='inferno') 
4. The resulting map shows no sign of the 5 degree increments.

I suspect that the error lies in the "_euler2axangle_signal(euler)" function by changing the return statement in this function from "return np.rad2deg(euler2axangle(euler[0], euler[1], euler[2])[1])" to return np.rad2deg(euler2axangle(np.deg2rad(euler[0]), np.deg2rad(euler[1]), np.deg2rad(euler[2]))[1]) I get the behaviour I expected, and an orientation map with 5 degree increments along x and y axes.

After I changed the "_euler2axangle_signal(euler)" function the orientation map looks like this:
[image]

dnjohnstone, 2019-11-14:
Thanks @EirikOpheim - yes I think that's probably a bug. Could you submit a bug fix PR for it?

@pc494 is that a part of the code you tweaked recently?

pc494, 2019-11-14:
It seems likely in my adjustments to the `.correlate()` method I dropped a `deg2rad` conversion somewhere.

The correction suggested seems best to me.

EirikOpheim, 2019-11-14:
I will try to submit a bug fix.

commit eaf76c0a (2019-11-14): Merge #529 fixing _euler2axangle()

closes #528
closed 2019-11-14 by pc494